randdusing/cordova-plugin-bluetoothle #125: Background BLE support

t2wu, 2015-01-19:
Hi this is a question rather than an issue.

There is a background mode for BLE support. But does the webview in either iOS or Android runs in the background? Suppose I hook up a heart rate monitor to it and I constantly receive data from it, does it work? I am been told from one source that with Cordova, web view in the background does not run and from another source that web view runninng in the background is not reliable. I tried it on iOS 7 and Javascript appears to run OK in the background. 

If web view doesn't work reliably, does the plugin provide some sort of native storage, and when the app jump back to foreground to fetch all the data at once? Just want to know more information about it, thank you.

randdusing, 2015-01-19:
I haven't tested this specifically, but the my heart rate app hasn't shown any issues with long readings yet.

t2wu, 2015-01-20:
No problems on both Android and iOS platform? That's good to know. Thanks. :)

closed 2015-01-22 by randdusing

t2wu, 2016-03-03:
Hi. On Android 4.x I have no problem with the plugin. On Android 5 and 6, I ran into issue where it is unable to read a characteristic after the device is connected in the background (Woke up by beacon, connect in the background, and attempt to read in the background).  The error message is "Unable to read". I tried both the version 2.x and 3.x of the Bluetooth plugin and both have the same problem on both Android 5 and 6. We did request user for permission via bluetoothle.requestPermission prior to putting the app into the background. Sniffer does not sniff any Bluetooth read operation (or perhaps we missed). What can be the problem?

randdusing, 2016-03-04:
I've been using the background mode plugin in my primary app, which may be why it works. Could you try that out? I'll still see if I can recreate the issue. What devices are you seeing this on?

t2wu, 2016-03-11:
Thank you so much! We're using HTC desire 816 ( Android 5) and HTC M9 ( android 6) and both exhibit the same behavior.

By the  plugin do you mean [cordova-plugin-background-mode](https://github.com/katzer/cordova-plugin-background-mode)?

t2wu, 2016-03-11:
I tried cordova-plugin-background-mode and it doesn't work. Digging into it I realized that the the reason it fails is that the a previous call to readCharacteristic() never gets completed -- onCharacteristicRead() wasn't called. So subsequent call to readCharacteristic was failing as a result.

randdusing, 2016-03-11:
Ahh yes, calling two read commands near the same time may cancel the first one on Android. After working on a few other projects, I can see the benefit of a queue for Android and now have plans to add it.