randdusing/cordova-plugin-bluetoothle #127: Bluethooth Scan Issues on Note 2 (android 4.3)

gilbertngandu, 2015-01-20:
Hi, 
I am having issues when using bluetoothle.startScan(startScanSuccess, startScanError, paramsObj);
I get "status":"started" but i never get a result. I can see that it is scanning when i call bluetoothle.isScanning(isScanningSuccess); Please help. I need to get the RSSI of available devices without connecting to them. I was hoping to use the result of the scan.
Thx

randdusing, 2015-01-20:
Scanning is a bit buggy in Android 4.3 on Samsung devices from what I can tell. I borrowed a Samsung Note II a few weeks ago and I believe this is what I encountered... If the device is paired, it will not show up in the scan. If the device isn't paired, it will show up in the scan. My next version will "fix" this by providing a retrieveConnected method for Android devices to return paired devices, but unfortunately I'm unable to figure out how to reliably perform service UUID filtering on it. It's actually in this version, but commented out and not working completely. Let me figure out how to add a test branch, and I'll release it shortly.

ipstas, 2015-01-28:
I can confirm the same issue on LG G3 4.4.2
I am trying to use it  under meteor. I can start scan (log says scanStarted, no error), but I have no results from the scan. 
And it is not about paired devices, there is brand new unpaired arduino ble on my table, evothings app can see it on the same phone.

randdusing, 2015-01-29:
Was evothings still connected to your device? I know my heart rate monitor stops advertising when connected to.

randdusing, 2015-02-27:
Still having issues? I've added retrieveConnected (pairing) support for Android in 2.1.0

ipstas, 2015-02-27:
No, in fact, that was the issue with meteor, it is not that easy to make cordova working for meteor framework. So I have finally managed to make it working, will probably create a package for meteor using your cordova one

You can see it - ready for to be installed on android https://github.com/XLazz/ble-meteor-andr
But it is only android 4 and up (cordova for meteor limitation)

Thanks again

randdusing, 2015-02-27:
I'm not familiar with Meteor. Does it run on top of Cordova or is it separate?

ipstas, 2015-02-27:
It includes cordova as a framework for mobile platforms. But can run in the pure browser as well. Not in the case of BLE of course
So my intent was to make BLE working for our app on android. And app is done with meteor with cordova on android/ios

randdusing, 2015-02-27:
Hmm, it sounds Meteor shouldn't be affecting the plugin code if Cordova is still being used.

ipstas, 2015-02-27:
no, it doesnt. The problem is in the wrapping. Meteor is async on a client side, sync on a server side, my issue was exactly about making it working async

randdusing, 2015-02-27:
On server side?

ipstas, 2015-02-27:
cant make it on a server side, server has no idea about client capability, it sits somewhere in a cloud

ipstas, 2015-02-27:
btw, i have upgraded android to 5.0.1, plugin still works, no issues

randdusing, 2015-02-27:
Ok, maybe I'm confused... there isn't any longer an issue with the plugin?

ipstas, 2015-02-27:
' cause I figured out that async issue. (Sorry, too old to think in terms of nonlinear programming, has started with fortran too long ago)

randdusing, 2015-02-27:
Ahh ok great. The asynchronous behavior of plugins was (still is) confusing for me as well.

closed 2015-02-27 by randdusing