randdusing/cordova-plugin-bluetoothle #129: Windows Phone 8

mismej, 2015-01-25:
In Life Cycle there is description in 4. how to use plugin on Android and iOS. How yo use it on Windows Phone 8?

randdusing, 2015-01-25:
Maybe @MiBLE can comment. I haven't verified the current Windows code. I started to work on Windows Phone 8.1, but I'm waiting until it's easier to debug Cordova based applications.

MiBLE, 2015-01-27:
Hello I don't use it in Windows Phone 8. I started project on C#. It was possible write something BLE to work only if upgrade cordova wp8 project to wp8.1.

randdusing, 2015-01-27:
Ahh ok, I didn't think Bluetooth LE was enabled on WP8, but the source folder was wp8. @MiBLE , your code only works for Windows 8 (desktop) apps?

MiBLE, 2015-01-27:
I was tested on wp8.1. I didn't on win8. Started with cordova 3.6.0  base code is written with JS.  There is no C# . I stopped to develop this project on C#.  How I see there is new version of cordova.  The way which I use: install visual studio 2013 express on win8.1 which running on VM ->install cordova-> add wp8 ->retarget to wp8.1->add plugin -> build with VS.

randdusing, 2015-01-27:
Were you able to debug on a device?

MiBLE, 2015-01-27:
Only C# code. For script I have used  console.log() and alert().

randdusing, 2015-01-27:
Thanks for the information. I'm going to take another look.

MiBLE, 2015-01-29:
I installed new version of Cordova via npm. it is 4.2.0. Create project - > Open it in VS 2013 Express with SDK 8.1 installed -> Retarget  to Windows Phone Silverlight 8.1 -> add plugin from https://github.com/MiBLE/BluetoothLE. It works without  any addition changes to done  like for Cordova 3.6.0.  C# code can be debugged.  Here example project https://github.com/MiBLE/BluetoothLE-VSProject

Timminater, 2015-02-19:
any updates on the windows phone 8 development? as windows phones (with bluetooth LE) are getting cheaper and cheaper (€60 in the Netherlands for a prepaid lumia 435, €80 for no contract/simlock)

randdusing, 2015-02-20:
I'm waiting to release the next version of my app. Then I'll start working on it assuming MiBLE's method works for me.

thomasbaggio, 2015-08-20:
@MiBLE @randdusing in plugin.xml I see that wp8 is commented how can I install this plugin for wp8 using cordova ? Thanks a lot

randdusing, 2016-03-24:
@thomasbaggio I was never been able to set up a working environment for the wp platform (Windows Phone 8). I'll be getting a Surface Pro in the next few days, but I'll be focusing my efforts on windows platform (8.1, 10, Phone 8.1) instead.

closed 2016-03-24 by randdusing