randdusing/cordova-plugin-bluetoothle #130: Phonegap Build 2.0.0 isConnected works for Android but not IOS

diverJohn, 2015-01-28:
Rand,

I'm currently using Phonegap 3.6.3 and trying to upgrade from your plugin version 1.0.6 to 2.0.0 and ran into a problem using the 'isConnected' function.   The 'isConnected' worked for both Android and IOS in version 1.0.6 but in version 2.0.0 it returns false for IOS when connected.

Code snippet:
function BluetoothLoop()
{
    var paramsObj = {"address":btAddr};    // btAddr filled from scan.
	bluetoothle.isConnected( isConnectedCallback, paramsObj );
}

function isConnectedCallback(obj)
{
    PrintLog(10, "BT: isConnectedCallback: " +  JSON.stringify(obj) );
}

Android (4.4.2):
Scan result: {\"advertisement\":\"AgEGEQYbxdWlAgAttOMRAPBgCtZIB/8AAQECJSkAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=\",\"status\":\"scanResult\",\"address\":\"D0:5F:B8:44:4A:E1\",\"rssi\":-55}

isConnected callback result: {\"address\":\"D0:5F:B8:44:4A:E1\",\"isConnected\":true}

IOS (7.0.2):
Scan result: {\"status\":\"scanResult\",\"name\":\"Nextivity Bridge\",\"advertisement\":\"AAEBAiUp\",\"address\":\"1CFBDB83-B643-F9DC-1425-493C4A549F64\",\"rssi\":-53}

isConnected callback result: {\"isConnected\":false,\"name\":\"Nextivity Bridge\",\"address\":\"1CFBDB83-B643-F9DC-1425-493C4A549F64\"}

Thanks,
John

mpedersen15, 2015-02-23:
I'm seeing this behavior as well. Though, when my BLE device IS NOT connected on iOS, isConnected is returning TRUE. So in the short term, it looks like you could do platform specific callbacks that flip flop the true and false?

randdusing, 2015-02-26:
Not sure how this one got by... Will be fixed shortly.

randdusing, 2015-02-27:
Should be good to go now.

closed 2015-02-27 by randdusing