randdusing/cordova-plugin-bluetoothle #131: Phonegap Build v1.0.6 writeSuccess works for Android but not IOS

diverJohn, 2015-01-29:
Rand,

I'm currently using Phonegap 3.6.3 and your plugin version 1.0.6.   Whenever I write data using the "noResponse" feature I get a writeSuccess callback for Android but not IOS.  I think that your documentation implies that there should be no callback for "noResponse" but I get one for Android.

Code snippet:
var paramsObj = {"value":u64, "serviceUuid":bridgeServiceUuid, "characteristicUuid":bridgeRxCharacteristicUuid, "type":"noResponse"};
bluetoothle.write(writeSuccess, writeError, paramsObj);

 Thanks,
John

randdusing, 2015-01-29:
I haven't been able to well test the write functionality with my devices. It looks other people have experienced this on Android. http://stackoverflow.com/questions/21791948/android-ble-gatt-characteristic-write-type-no-response-not-working

What version of Android are you running on?

diverJohn, 2015-01-29:
All 3 of our Android test platforms behave the same.
Samsung S3:  Android 4.4.2
Samsung Note 8.0:  Android 4.4.2
Nexus 5: Android 4.4.4.

We are developing our BLE hardware based on the TI CC2541.   I have several CC2541 eval boards and would be happy to send you one to help you investigate the write functionality.

randdusing, 2015-01-30:
I'm traveling around the world until at least November, so it will be hard to send something. :) Where are you located? I couldn't get the emulator working on my Macbook. If it's fairly cheap, I could pick up a device that can be written to.

diverJohn, 2015-01-30:
Traveling around the world until November...sounds great.   We are in San Diego, Ca.

randdusing, 2015-01-30:
If I bought one at the store, would I need to program so logic or can I
plug it in and begin interacting with it via Bluetooth?

diverJohn, 2015-01-30:
Here is a link to the eval kit that I was going to send you: http://www.ti.com/tool/cc2540dk
Here is a link to the documentation: http://www.ti.com/lit/ug/swru301a/swru301a.pdf
There are two eval boards included and both come pre-programmed on a newly purchased system.  One of the two systems comes pre-programmed as a slave device so you can start talking to it right away.   The BLE software stack comes with many example programs that you can compile and run.   The downside is that it takes the IAR 8051 compiler to build the code.    The IAR compiler can be downloaded and used for free for 30 days or purchased, however the cost is a couple of thousand.
The kit also includes the USB dongle sniffer module which can be used to sniff the air and read the BLE packets.

randdusing, 2016-03-24:
iOS will now return a success callback when writing without response as of 4.0.0-beta.1

closed 2016-03-24 by randdusing