randdusing/cordova-plugin-bluetoothle #132: Blood pressure indicate

MHrezdovic, 2015-02-02:
Hi guys, 

i have a issue i would like to read data from blood pressure meter but somehow im unable to to get anything from the BP meter. After subscribing to BP monitor i always get 0 or error. Someone have an ide what is the life cycle of BP measurement. e.g.: Discover connect subscribe read disconect close.

Or if someone made some similar app can you provide the source code?

thx

randdusing, 2015-02-03:
What device, services and characteristics are you using?

MHrezdovic, 2015-02-03:
I have  Medisana BU 550 Connect it has Blood pressure service 1810 and
blood pressure characterestic 2A35. From what i learned the Blood pressure
measurement should use indicate so the client(smartphone) should somehow
respond with ack to the server(BPM) but after i subscribe the client to BP
service and characteristic i get nothing.

randdusing, 2015-02-03:
Are you getting any returned data value? And how are you parsing it?

MHrezdovic, 2015-02-04:
Im not getting anything i somehow cant figgure out how. I tryed to read the
characteristic but they seems to expect some command to start transmiting
the data

randdusing, 2015-02-04:
Check out the example about subscribing to a heart rate monitor. Blood pressure appears similar and uses the client configuration characteristic. Main difference seems to be notify vs indication.

https://developer.bluetooth.org/gatt/services/Pages/ServiceViewer.aspx?u=org.bluetooth.service.blood_pressure.xml

MHrezdovic, 2015-02-04:
Yes, i figure that out but i am clueless about how the indication works. I
believe that on indicate client needs to send some message or some
acknowlidgement to the server. But cant find anywhere what this ack should
be.

randdusing, 2015-02-04:
Ahh, I see what you mean. I don't see anything in the API about acknowledging indications, so I assumed it was something the Bluetooth stack handled.

MHrezdovic, 2015-02-04:
Well you might be right but i did not come accross some file that would
shed some light into this topic :(

randdusing, 2015-02-27:
Were you able to resolve your issue?

MHrezdovic, 2015-02-27:
Hi,

No im somehow unable to acknowlidge the transmition so now luck so far...

randdusing, 2016-01-24:
Still having issues? I wouldn't mind testing it on a physical blood pressure device, but it looks like the Medisanas are pretty expensive to obtain in the US. Do you know of any other brands?

MHrezdovic, 2016-01-25:
yup, but i didnt gave it much time afterwards...never managed to
acknowlidge the transmition

randdusing, 2016-03-31:
I tested this out with the server side code I wrote. I created a characteristic with the notify property and one with the indicate property. Using another iOS device, I was able to successfully subscribe to both characteristics.

On Android, I could subscribe to the characteristic with notify, but not with indicate. I then tried again by passing isNotification => false within the subscribe params, and it worked. By the way, I'm changing subscribe to just automatically detect whether the notify or indicate property is set, so that the app doesn't need to specify isNotification anymore.

So with the change that will be committed to the master branch within an hour or two, there's nothing that needs to be done at the app level to handle indications. Let me know if this doesn't work.

closed 2016-03-31 by randdusing