randdusing/cordova-plugin-bluetoothle #134: Changing Connection Priority on Android

kgillespieanaren, 2015-02-06:
Working on a patch to allow the BluetoothLE plugin to change the Connection Request priority. Apparently you can improve the throughput by request a high priority.

I have created a patch file for the master branch.

http://pastebin.com/mcXbaqpH

Haven't tested this but it's based on some code I was doing for a native android app. This wouldn't be applicable to iOS but it could be useful for Android.

kgillespieanaren, 2015-02-06:
Should not be optInteger not optString

randdusing, 2015-02-07:
I'm not too familiar with patch files. Can it be turned into a pull request?

kgillespieanaren, 2015-02-08:
I'll let you know when my fork is tested and I can confirm that this patch works.

randdusing, 2015-02-08:
Thanks, it seems low risk for existing users, so it should be good after your testing.

kgillespieanaren, 2015-02-09:
It will require that users increase their android-minSdkVersion to at least 21.

kgillespieanaren, 2015-02-09:
I was wondering if it would make sense to break out the functions that are specific to each platform into their own javascript object, and then write wrapper handlers that try and make the best decision based on the platform.

So you would have bluetoothle.android.requestConnectionPriority(...)

kgillespieanaren, 2015-02-09:
I added a check so that if the user is not using API 21 or greater the code simply returns. Although now I have a problem with API 21 where I cannot write to characteristics... Oh boy.

kgillespieanaren, 2015-02-09:
Looks like API 21 on Android is causing a lot of regressions in the plugin, also Phonegap doesn't officially support it

Wondering if the new Bluetooth LE stuff has something to do with it,
https://developer.android.com/reference/android/bluetooth/le/package-summary.html

randdusing, 2015-02-09:
Oh, I wasn't aware of all the new Bluetooth functionality. My app seems to work fine on API 21, but I'm not writing any characteristics. Eventually I'll probably need to completely rewrite it for API 21 and then allow the user to decide which "code path" to use. But I'm not sure if that's easily possible with the cordova plugin system...

I'm not sure if I see the value in making separate handlers for each platform... If the function doesn't apply, it can be ignored or send a method not supported error back.

kgillespieanaren, 2015-02-10:
I figured out the problem, it was my end and Android not handling a message queue well. My own messaging queue tried to double up a message. My code seems to work, though I am not able to claim any changes to performance yet.

randdusing, 2015-02-10:
Even on API 21, you mean?

kgillespieanaren, 2015-02-10:
Yes on API 21, I need to run my benchmarking on it and haven't had time today. I'll hopefully get an idea soon. I am right now debugging some unrelated bugs that have occured do to the switch to API 21, apparently the Webkit stuff handles cookies and authentication a little differently and this is a bigger issue for me.

randdusing, 2015-02-27:
This has been added to the master branch. I made a few changes with error handling.

closed 2015-02-27 by randdusing