randdusing/cordova-plugin-bluetoothle #135: WriteError callback isn't being called (sometimes)

Danyc0, 2015-02-11:
Hi,
Sometimes, when the device i'm connecting to reboots unexpectedly whilst i'm writing to one of it's characteristics, the WriteError callback isn't called. Sometimes it works, sometimes it doesn't, it seems random. I'm using Phonegap 4.1.2 and plugin version 2.0.0 (is there a more up to date version?),

Thanks,
Dan

Danyc0, 2015-02-11:
I've just checked, and the WriteSuccess callback isn't being called either (not that it should). The connection dies, but it seems to sometimes just stop writing, rather than throwing the Error Callback. 
Thanks again,
Dan

Danyc0, 2015-02-11:
I think it's something to do with when the reset happens during a write (rather than just before), so the write itself actually fails, rather than any checks that might happen before the write

randdusing, 2015-02-11:
Does it ever write successfully? Perhaps the device is crashing when receiving a bad command. What type of device? And how are you encoding the data to send?

Danyc0, 2015-02-13:
It's a BLE113 board from BlueGiga, and an iPhone 4.
Yes, it writes perfectly fine whilst in normal operation. I've been testing sending large amounts of data in small packets (100kB in blocks of 13 bytes) and that also works perfectly. However, when i manually reset the board midway through the data transfer (which i've been doing to test how my software recovers), sometimes the WriteError callback is called, but sometimes it's not.
The data is just a fixed set of ASCII values which i run through the "bytesToEncodedString()" method.

Example transfer of how I'd be sending 52 bytes (in the form of ASCII characters) in 13 byte blocks, each block being a separate call to the "write()" method:

ABCDEFGHIJKLM ABCDEFGHIJKLM ABCDEFGHIJKLM ABCDEFGHIJKLM

randdusing, 2015-02-13:
When you don't receive a write error status, do you receive a disconnected status or nothing?

Danyc0, 2015-02-13:
sorry, what do you mean by "disconnected status"?

randdusing, 2015-02-13:
When you manually reset the device, it should be disconnecting from your phone.

Danyc0, 2015-02-16:
Am i right in saying there isn't a callback available for when it disconnects unexpectedly?

randdusing, 2015-02-16:
A write or disconnect callback? If you turn off the device, I'd still expect a disconnect callback, but not necessarily a write callback.

Danyc0, 2015-02-16:
This might be me being stupid, but i can't find any mention of a disconnect callback anywhere?

Danyc0, 2015-02-16:
Not sure if the disconnect callback is called, i will check, thanks

randdusing, 2015-02-16:
Check out: https://github.com/randdusing/BluetoothLE#connect
If the device disconnects unexpectedly, it will use the connect callback

randdusing, 2015-02-27:
Were you able to resolve your issue?

Danyc0, 2015-02-27:
Yes, thank you! sorry for the late reply

closed 2015-02-27 by Danyc0