randdusing/cordova-plugin-bluetoothle #137: Did anything change with respect to discovery?

mrquincle, 2015-02-18:
Has there been any changes in https://github.com/randdusing/BluetoothLE#discover? I have been using a version that didn't need arguments several months ago. Does it now require an argument? Do I set `address` in `paramsObj` to `[]` if I am already connected to a device? Or did anything change in the sense that you are now not allowed to be connected and discovery takes care of that automatically? I don't understand namely why there is an argument if you can only discover services with connected devices...

```
D/CordovaLog( 5622): file:///android_asset/www/js/ble.js: Line 200 : Beginning discovery of services
D/CordovaLog( 5622): file:///android_asset/www/js/ble.js: Line 380 : Discover error: connect - No device address
```

Thanks a lot!

randdusing, 2015-02-18:
As of version 2.0.0, you can connect to multiple devices simultaneously and most functions now require you to specify the device (via address).

mrquincle, 2015-02-18:
Clear! Works indeed. Still an update to the README.md would be nice. :-)

randdusing, 2015-02-18:
The readme should be up to date...at least discover mentions the address requirement. The next update will include a change log.

mrquincle, 2015-02-18:
It states `bluetoothle.discover(discoverSuccess, discoverError);` without additional argument.

randdusing, 2015-02-19:
Ahh, I need to fix that. If you scroll down a little further, it mentions the parameters.

mrquincle, 2015-02-19:
Yes, I discovered after your remark. :notes:

randdusing, 2015-02-27:
Documentation has been updated.

closed 2015-02-27 by randdusing