randdusing/cordova-plugin-bluetoothle #138: How to scan ?

przmak, 2015-02-19:
Got some issue with scanning

    function initializeBT() {
    var params = {request: true, statusReceiver: false};
    bluetoothle.initialize(function (status) {
        //initializaded
        bluetoothle.isEnabled(function(status){
            if(!status.isEnabled){
                //Disabled
                $('#deviceListCon').html("Bluetooth is disabled");
                return;
            }          
            var uids = {serviceUuids : []};
            bluetoothle.startScan(function (status) {
                //Success
                $('#deviceListCon').html($('#deviceListCon').html() + JSON.stringify(status) + "<br/>");
                // Allways getting  "Status":"SCANSTARTED"
                
            }, function (error) {
                //Error
                $('#deviceListCon').html("error = " +  JSON.stringify(error));
            //},uids);
            //},[]);
            },null);
        });        
    },function(){
        //error
        $('#deviceListCon').html("Error in initialization, is your bluetooth turn on ?  <br/>");
    },params);
    }

    $('#sendBTInfo').on("tap", function () {
    initializeBT();
    
    });

I'm allways getting STATUS: SCANSTARTED as a result of success function, how can I list all the devices in the range? or at least one at a time, but not specific one? Dont have any spec info about the devices

blhack, 2015-02-20:
You should eventually get a status to your success callback of "scanResult".  Are you sure that something is definitely in range?  Check out the application "lightblue" (available on the mac app store).  It's good for debugging.

The status.name and status.address properties will be the things that you will care about.

My code has an object called devices (devices = {}) that gets populated during each time a status of "scanResult" is indicated to the success callback function.

That is how I keep track of the results.  When stopScan is called, the contents of devices is written out ot the screen.

przmak, 2015-02-20:
Yeah, I'm sure, there are some devices in the range
My build in bluetooth scanner is showing me other devices in the range

przmak, 2015-02-20:
I just got a briliant thought , 
Does other devices need to be api19 + as well ?
That would solve everything : D 

My main is api19 but the other devices are below 18

randdusing, 2015-02-20:
Bluetooth or Bluetooth LE Scanner? Normal Bluetooth devices won't show up in the scan. The other devices most likely need to be API18+ or they won't support Bluetooth.

randdusing, 2015-02-27:
Were you able to resolve your issue?

closed 2015-02-27 by randdusing