randdusing/cordova-plugin-bluetoothle #139: Can't disconnect a device once it goes out of range

blhack, 2015-02-20:
If I connect to a device, and then the device goes out of range (or if I power the BLE device off), I am unable to disconnect from it.  This means that I am also unable to reconnect, connect, etc. to it.

disconnectSuccess is called with status "disconnecting", but then never achieves "disconnect" status, and disconnectError is never called.

(Is this normal behavior?  Or is there something else I need to call so that I am able to reconnect to devices if they come back into range?)

I am on a Nexus 5 - Android 5.0.1.

randdusing, 2015-02-20:
This shouldn't be normal behavior. Is it possible the code handling the disconnecting status is causing an error?

I just upgraded my Nexus 7 to Android 5 a few weeks ago and can verify the behavior when I test the latest version of my app.

blhack, 2015-02-20:
I am able to reproduce the error using your example code.  To reproduce, do the following:

1) Initialize 
2) connect / discover a device
3) Subscribe to a service
4) disconnect power from the BLE device (simulating an out of range event)
5) observe "connect error" - "connection failed"

When you try to "disconnect" from the device at this point, this is the series of events:

Disconnect : {"address":"F0:13:B7:8B:7B:42"} index.js:581
Disconnect Success : {"address":"F0:13:B7:8B:7B:42","name":"RedBear","status":"disconnecting"} index.js:590
Disconnecting

Even after power is reapplied to the BLE device, disconnect/reconnect events don't work.  The error is "isNotDisconnected" when trying to reconnect, or the previous thing when trying to disconnect (the disconnectSuccess function is only called once.

randdusing, 2015-02-26:
I was able to reproduce the issue and believe it only affects 5.0+. Below 5.0 when a device was disconnected unexpectedly, the function returned status=>SUCCESS (Yes, this didn't really make sense) and state=>DISCONNECT. Now in 5.0, it's returning status=>NO_SUCCESS. My code to handle NO_SUCCESS was incorrect since I assumed this error would only occur on the initial connection attempt. Checking out some other bugs, and I should commit later tonight

randdusing, 2015-02-27:
Should be good to go now.

closed 2015-02-27 by randdusing

blhack, 2015-02-27:
You are the best!  Thanks a bunch!