randdusing/cordova-plugin-bluetoothle #140: Connecting to device with mac address

Gil401, 2015-02-23:
Hi @randdusing 
Is there a way to connect to a BT device with a mac address (and not with UUID)?

randdusing, 2015-02-26:
On Android, a MAC address is used.
On iOS, a UUID is used that changes between devices. For some reason they decided to hide the actual address.

What exactly are you trying to do?

Gil401, 2015-02-28:
Hi @randdusing , Thank you for the response.

I am trying to connect to a HC-06 BT device or other Android device

:This is the code snippet :

var address = { "address": "10:14:06:03:04:28" };
bluetoothle.connect( function () { alert("connetcted"); }, function () { alert("unable to connect"); },
					address);

but it doesn't connect.

randdusing, 2015-02-28:
Is the HC-06 BT device Bluetooth LE compatible?

Gil401, 2015-02-28:
I will recheck... 
should it work from Android to Android?

randdusing, 2015-02-28:
Sometimes :)  It depends if the Android is advertising Bluetooth LE, which I believe it usually doesn't unless specifically activated by some application.

closed 2015-10-19 by randdusing