randdusing/cordova-plugin-bluetoothle #141: isDiscovered returns null after reconnect

mpedersen15, 2015-02-24:
Hi Rand,

I'm working on creating some robustness when connecting to my BLE peripheral, and want to make sure I can reconnect and then write to the device after it gets disconnected. The isDiscovered function works perfectly after the initial connection to the peripheral, but after I reconnect successfully, isDiscovered only seems to return null. Could this be a bug of some kind?

Thanks for the help and the great plugin!

randdusing, 2015-02-26:
Thanks for the bug report. I've found the issue on Android and should be very similar on iOS. I should be wrapping up the next version tonight.

randdusing, 2015-02-26:
Oops forgot iOS discovery works much differently and this shouldn't be possible on iOS. Was your original bug report was only for Android?

mpedersen15, 2015-02-26:
Yep you're correct, only for Android. iOS has been working pretty consistently for me! Thanks for checking this out!

randdusing, 2015-02-27:
Should be good to go now.

closed 2015-02-27 by randdusing

mpedersen15, 2015-02-27:
Awesome thanks!