randdusing/cordova-plugin-bluetoothle #144: Project will not build with this plugin installed

blhack, 2015-02-27:
To replicate, run the following commands:

cordova create hello com.example.hello HelloWorld
cd hello/
cordova platform add android

(At this point cordova build will execute successfully)

cordova plugin add https://github.com/randdusing/BluetoothLE

(At this point cordova build will fail with the following output):

    [javac] Compiling 3 source files to /private/tmp/hello/platforms/android/ant-build/classes
    [javac] /private/tmp/hello/platforms/android/src/com/randdusing/bluetoothle/BluetoothLePlugin.java:1736: cannot find symbol
    [javac] symbol  : variable CONNECTION_PRIORITY_BALANCED
    [javac] location: class android.bluetooth.BluetoothGatt
    [javac]     int androidPriority = BluetoothGatt.CONNECTION_PRIORITY_BALANCED;
    [javac]                                        ^
    [javac] /private/tmp/hello/platforms/android/src/com/randdusing/bluetoothle/BluetoothLePlugin.java:1752: cannot find symbol
    [javac] symbol  : variable CONNECTION_PRIORITY_LOW_POWER
    [javac] location: class android.bluetooth.BluetoothGatt
    [javac]       androidPriority = BluetoothGatt.CONNECTION_PRIORITY_LOW_POWER;
    [javac]                                      ^
    [javac] /private/tmp/hello/platforms/android/src/com/randdusing/bluetoothle/BluetoothLePlugin.java:1756: cannot find symbol
    [javac] symbol  : variable CONNECTION_PRIORITY_BALANCED
    [javac] location: class android.bluetooth.BluetoothGatt
    [javac]       androidPriority = BluetoothGatt.CONNECTION_PRIORITY_BALANCED;
    [javac]                                      ^
    [javac] /private/tmp/hello/platforms/android/src/com/randdusing/bluetoothle/BluetoothLePlugin.java:1760: cannot find symbol
    [javac] symbol  : variable CONNECTION_PRIORITY_HIGH
    [javac] location: class android.bluetooth.BluetoothGatt
    [javac]       androidPriority = BluetoothGatt.CONNECTION_PRIORITY_HIGH;
    [javac]                                      ^
    [javac] /private/tmp/hello/platforms/android/src/com/randdusing/bluetoothle/BluetoothLePlugin.java:1776: cannot find symbol
    [javac] symbol  : method requestConnectionPriority(int)
    [javac] location: class android.bluetooth.BluetoothGatt
    [javac]     boolean result = bluetoothGatt.requestConnectionPriority(androidPriority);
    [javac]                                   ^
    [javac] Note: /private/tmp/hello/platforms/android/src/com/randdusing/bluetoothle/BluetoothLePlugin.java uses unchecked or unsafe operations.
    [javac] Note: Recompile with -Xlint:unchecked for details.
    [javac] 5 errors

    BUILD FAILED

randdusing, 2015-02-27:
Try setting the target sdk to 21

blhack, 2015-02-27:
This fixes it.  Thank you.

To anybody who lands here from google: from within your project directory, edit the file ./platforms/android/project.properties

And change the target to android-21

If you do not have the newest version of the android sdk installed, run the command "android" on your terminal, and install it using that utility.

Thanks, randdusing!

randdusing, 2015-02-27:
Glad it worked out!

closed 2015-02-27 by randdusing