randdusing/cordova-plugin-bluetoothle #145: Android compiler error

xam8re, 2015-03-02:
-compile:
    [javac] Compiling 3 source files to /home/massimiliano/project/MyApp/platforms/android/ant-build/classes
    [javac] /home/massimiliano/project/MyApp/platforms/android/src/com/randdusing/bluetoothle/BluetoothLePlugin.java:1736: error: cannot find symbol
    [javac]     int androidPriority = BluetoothGatt.CONNECTION_PRIORITY_BALANCED;
    [javac]                                        ^
    [javac]   symbol:   variable CONNECTION_PRIORITY_BALANCED
    [javac]   location: class BluetoothGatt
    [javac] /home/massimiliano/project/MyApp/platforms/android/src/com/randdusing/bluetoothle/BluetoothLePlugin.java:1752: error: cannot find symbol
    [javac]       androidPriority = BluetoothGatt.CONNECTION_PRIORITY_LOW_POWER;
    [javac]                                      ^
    [javac]   symbol:   variable CONNECTION_PRIORITY_LOW_POWER
    [javac]   location: class BluetoothGatt
    [javac] /home/massimiliano/project/MyApp/platforms/android/src/com/randdusing/bluetoothle/BluetoothLePlugin.java:1756: error: cannot find symbol
    [javac]       androidPriority = BluetoothGatt.CONNECTION_PRIORITY_BALANCED;
    [javac]                                      ^
    [javac]   symbol:   variable CONNECTION_PRIORITY_BALANCED
    [javac]   location: class BluetoothGatt
    [javac] /home/massimiliano/project/MyApp/platforms/android/src/com/randdusing/bluetoothle/BluetoothLePlugin.java:1760: error: cannot find symbol
    [javac]       androidPriority = BluetoothGatt.CONNECTION_PRIORITY_HIGH;
    [javac]                                      ^
    [javac]   symbol:   variable CONNECTION_PRIORITY_HIGH
    [javac]   location: class BluetoothGatt
    [javac] /home/massimiliano/project/MyApp/platforms/android/src/com/randdusing/bluetoothle/BluetoothLePlugin.java:1776: error: cannot find symbol
    [javac]     boolean result = bluetoothGatt.requestConnectionPriority(androidPriority);
    [javac]                                   ^
    [javac]   symbol:   method requestConnectionPriority(int)
    [javac]   location: variable bluetoothGatt of type BluetoothGatt
    [javac] Note: /home/massimiliano/project/MyApp/platforms/android/src/com/randdusing/bluetoothle/BluetoothLePlugin.java uses unchecked or unsafe operations.
    [javac] Note: Recompile with -Xlint:unchecked for details.
    [javac] 5 errors

I've sdk 21 installed, The target it's sdk 19. 

I've got two old samsung android, if you need some test, 
Galaxy mini, Galaxy Ace, Galaxy Next

randdusing, 2015-03-02:
You'll need to set the target sdk to 21 in your project.properties file.

randdusing, 2015-03-02:
See #136

closed 2015-03-02 by randdusing

xam8re, 2015-03-02:
Ok, I've seen the update just a bit after the bug post. 
Isn't much restrictive API21 (Lollipop) ?

randdusing, 2015-03-02:
I'm not really sure how restrictive API21 is compared to the others. If you can't use API21, you can comment out the requestConnectionPriority code.

xam8re, 2015-03-02:
At the moment,, lollipop has 1.2% share.  Min API 21 means having Lollipop (android 5.0) terminal.
https://developer.android.com/about/dashboards/index.html

randdusing, 2015-03-02:
The Min API doesn't need to be changed, only the Target API. Setting the target API to 21 won't prevent the application from being installed on older devices.

http://developer.android.com/guide/topics/manifest/uses-sdk-element.html