randdusing/cordova-plugin-bluetoothle #146: 'subscribe' action on IOS returns instantly a 'subscribedResult', if Characteristic's value was previously set

donotor, 2015-03-13:
Hi there:

When i perform a 'subscribe' to a characteristic on IOS, if the value of the characteristic was previously set, i receive a 'subscribedResult' on the very moment with characteristic's value.
I don't think this is the right behavior, since the value of the characteristic didn't  change while subscribed.

On Android it works as expected: i 'subscribe' to the characteristic, and i won't receive a 'subscribedResult' until the value of the characteristic actually changes (is written).

I don't know how to code for IOS, otherwise i would try to pull a fix myself. Many thanks for this great piece of sw.

randdusing, 2015-03-14:
I'm not sure if I follow the process for recreating. Subscribe to a characteristic, unsubscribe, and then subscribe again? One the second subscribe, you immediately receive a subscribedResult before the subscribed event?

donotor, 2015-03-15:
No. Let me explain the process:

ANDROID (works as expected):
1. The machine with bluetooth writes value (a) on characteristic (A).
2. Android subscribes to (A) and gets the first notification: (subscribed).
3. Later, the machine writes value (b) on characteristic (A), and Android receives a (subscribedResult) with (b) value.

IOS:
1. The machine with bluetooth writes value (a) on characteristic (A).
2. IOS subscribes to (A) and gets the first notification: (subscribed), and inmediatly a second notification (subscribedResult) with (a) value.
3. Later, the machine writes value (b) on characteristic (A), and IOS receives a (subscribedResult) with (b) value

I think the right behavior is the one in Android, as i'm only expected to receive subscribedResults on value changes (or writes) WHILE i'm subscribed.

randdusing, 2015-03-15:
If I'm following your iOS steps, it sounds like you are subscribed when you get the first subscribedResult. On iOS step 2, are you getting a subscribedResult before subscribed?

donotor, 2015-03-15:
Yes, the (subscribedResult) on step 2 in IOS is right AFTER the (subscribed) notification.
But the nature of the notification should be to receive it upon updates of the value, and no update of the value happens on step 2...
It breaks somehow  a request/response system I want to develop over the subscription system. I expect notifications when char's value changes, and that happens after i make a "question" in certain way.
As i said, on Android it works differently, and (in my opinion), as expected.

randdusing, 2015-03-15:
Ahh ok, I follow what's happen. I'll see if I can reproduce with my heart rate monitor. What type of characteristic are you subscribing too?

donotor, 2015-03-15:
It's a custom characteristic for a machine we are developing with bluetooth. This characteristic is meant to transmit "responses" upon request received on other characteristic, in a transactional way...

donotor, 2015-03-16:
I found this discussion:

http://stackoverflow.com/questions/28245090/will-ble-peripherals-send-a-characteristics-value-automatically-upon-subscribin

The first answer states what i'm saying. I'm trying to understand where in the code of BluetoothLEPlugin.m it performs a read right after subscribing, but i don't see it...

randdusing, 2015-09-26:
Still having this issue?

randdusing, 2016-01-24:
Closing due to inactivity. Reopen if needed

closed 2016-01-24 by randdusing