randdusing/cordova-plugin-bluetoothle #147: Example script doesn't work

nonozor, 2015-03-15:
Hello,

Here is the error I've got on Eclipse Log, running the example script  :
Line 267 : Uncaught TypeError: Cannot read property 'initialize' of undefined
[INFO:CONSOLE(267)] "Uncaught TypeError: Cannot read property 'initialize' of undefined", source: file:///android_asset/www/js/index.js (267)

Other error (but nearly the same):
 file:///android_asset/www/js/index.js: Line 355 : Uncaught TypeError: Cannot read property 'startScan' of undefined
[INFO:CONSOLE(355)] "Uncaught TypeError: Cannot read property 'startScan' of undefined", source: file:///android_asset/www/js/index.js (355)

randdusing, 2015-03-15:
Did you install the actual plugin?

nonozor, 2015-03-15:
Yes of course

randdusing, 2015-03-15:
Ok, just checking :)

And is var bluetoothle; still at the top? Can you try printing out the statement to see if it's getting populated? It should be populated by bluetoothle.js

nonozor, 2015-03-15:
Ok, i think I see what's wrong, bluetoothle.js is not imported in index.html nor in index.js..

nonozor, 2015-03-16:
Now it works fine after adding the line "<script type="text/javascript" src="bluetoothle.js"></script>" just before "<script type="text/javascript" src="js/index.js"></script>" in index.html on line 13.

Great job, this is just what I was looking for, thank you for that plugin.

randdusing, 2015-03-16:
Hmm, it shouldn't need to be added like that. Let me see if I can recreate the issue.

nonozor, 2015-03-16:
Maybe it's because I've started from existing Cordova Android project in
Eclipse, I've just copy js and html files in the asset directory, before
that I've imported the plugin Bluetootle by the command line "cordova add
plugin".

randdusing, 2015-03-16:
What do you mean by existing project exactly? I just verified on iOS, but I did get a crash on an emulated Android device. Let me pull out my physical device and see what the issue is.

nonozor, 2015-03-17:
By "existing project", I mean that I used a different Cordova project that
had ever been generated and imported into Eclipse, I've just replace the
files (html+js+css) in the assets/www directory.

runaC, 2015-04-24:
Hello,

I am new to cordova.I am trying to run the example with the plugin and I am also getting the same issue.I have created a fresh cordova project .Copied  the WWW folder from example and pasted it my created project .Not sure if this is an issue .Even adding the bluetoothle.js is not working for me.Internally  bluetoothle.js is referring to cordova.js where it fails .I guess including bluetoothle.js explicitly is not a good way because it should be fetched from plugin folder.Need your help to figure out where did I go wrong in running this example.
Thanks
RC

runaC, 2015-04-26:
Hi,
You can ignore my previous message.Finally it worked for me on actual device.Thanks for such an awesome plugin.

Thanks,
RC

randdusing, 2015-06-09:
Anyone still having issues?

timtim1, 2015-06-13:
Yes, I am having that exact same issue.
Other error (but nearly the same):
file:///android_asset/www/js/index.js: Line 355 : Uncaught TypeError: Cannot read property 'startScan' of undefined
[INFO:CONSOLE(355)] "Uncaught TypeError: Cannot read property 'startScan' of undefined", source: file:///android_asset/www/js/index.js (355)

Except, I get the same type of errors for any cordova project I try to build and emulate.

soumendub3, 2015-09-06:
Hi Rand,

I did exactly what runaC's message says and I cannot get it to work. I am using PhoneGap to test my app. I see that runaC said that the issue was resolved but I cannot - I see all the buttons, but none of the buttons do anything for me. Can you help? 

@runaC how did your issue got resolved? Can you post the details please?

Thanks!!

runaC, 2015-09-06:
Hi,
My issue got resolved using the actual device.The emulator did not work for
me.

Thanks!
RC

randdusing, 2015-09-26:
Reopen if you continue to have any problems. Next minor update will have a note about not using the emulator for testing in the documentation.

closed 2015-09-26 by randdusing