randdusing/cordova-plugin-bluetoothle #148: Android Service UUID scan filtering no longer works.

kgillespieanaren, 2015-03-16:
Not sure what happened between the resent versions but it appears that on all my Android devices their ability to use the scan filtering doesn't work.

When I try to do the following with the correct Service UUID 

```
bluetoothle.startScan(baseApp.scanSuccess, baseApp.scanFailed, {"serviceUuids": [app.serviceUUID]});
```

Where app.serviceUUID is the primary 128 bit service uuid that is being broad casted in my advertising data. This scanning code works fine on iOS devices but does not work on Lollipop, Kitkat on a verity of Android system, Nexus 7/9, Asus Memo, Galaxy Tab, ect...

I have the following work around for my Android devices but it's a little bit of a kludge. I simply use this code.

```
bluetoothle.startScan(baseApp.scanSuccess, baseApp.scanFailed, {"serviceUuids": []});
```

And I filter the raw service UUID out of the advertising data.

```
if (device.platform == androidPlatform) {
	
	var advData = retObject.advertisement;
	
	var rawData = atob(advData);
	
	serviceUuid = "";
	
	for(var i = 20; i >= 5; i--) {
		serviceUuid = serviceUuid + ("00" + rawData.charCodeAt(i).toString(16)).slice(-2);
		
		if(i == 17 || i == 15 || i == 13 || i == 11) {
			serviceUuid = serviceUuid + "-";
		}
		
	}
	
	if(serviceUuid !== app.serviceUUID) {
		return;
	}
}
```

Wondering if anyone else has seen this issue.

I have tried updating my Android SDK and still no joy. API version 22 targeting 19.

randdusing, 2015-03-16:
What's the actual 128bit UUID you're using? I'll look into it.

kgillespieanaren, 2015-03-25:
Here is an example

"0f99f9bf-9ae1-49a4-ad70-c79f4260983a"

ArMouReR, 2015-05-21:
I am have the same issue. There is any update ???

Chris1234567899, 2015-05-27:
How many filters are you using?
I had a similar problem when I used more than 13 filters, which was caused by an android bug in versions below Android 5.02. Try it with 5.02 or higher, it might help (remove the targeting to API 19).

hervethouzard, 2015-09-02:
Hello,

In my case filtering on iOs is not running.

My Service Id is something like this:
23092531-3A42-5A5D-9862-91F01BA1D866

randdusing, 2015-09-26:
I noticed this on my app as well. I also noticed there's a new scanning API with API21. I'll see if I experience the same problems using this.

http://developer.android.com/reference/android/bluetooth/le/BluetoothLeScanner.html

randdusing, 2015-09-26:
Actually I don't see the issue on my Nexus 7 running Android 5.1.1. It may have been my co-worker who saw the issue.

randdusing, 2016-01-24:
A pull request was just added to support API21 scanning. I'm going to make a few changes and this should be available within a few a days. I'm curious if this helps the issue.

randdusing, 2016-01-25:
New scanning is added. Let me know if that doesn't fix it.

closed 2016-01-25 by randdusing