randdusing/cordova-plugin-bluetoothle #149: bluetoothle.rssi never gets callback or error? (iOS)

matthewkremer, 2015-03-26:
So after connecting to a device and getting the status of "connected" I try booting off a call to ``bluetoothle.rssi()`` with the address of the device and nothing happens.

No error, no success callback, no error callback, nothing.

Any ideas? Thanks!

matthewkremer, 2015-03-27:
Interesting...I thought this was due to the problems with the callback in iOS 8.0 so I changed some code, wouldn't compile, replaced the ``.m`` file with the original, and it worked...

Now I'm just stumped, oh well! At least it's working :)

closed 2015-03-27 by matthewkremer

randdusing, 2015-03-27:
Glad, you got it figured out. Let me know if it reoccurs!

ArMouReR, 2015-06-24:
I have strange problem IOS8 - on one device everything works without problem.
On another device: the same problem as described here: no callback and no error...
Any idea how can I debug it ?

Thanks...