randdusing/cordova-plugin-bluetoothle #159: Writing characteristics larget than ATT_MTU

Tomino2112, 2015-06-02:
Hi there,

I have done a lot of research about writing longer values into characteristics, and if I understand it correctly it comes down to that ATT packets can be max size of 23 bytes - this is the limitation of the radio signal ATT is using and there is no physical way around, BUT there is software way (yay!)

According to specs and all sorts of discussions, SDKs (both on GATT client and server) are able to split and connect ATT packets and therefore enable reading/writing longer values. 

It seems pretty straight forward, and I would love to see it implemented in this plugin - any way you would have the time for it? 

Some resources:

https://community.freescale.com/message/441547#441547
https://www.safaribooksonline.com/library/view/Getting+Started+with+Bluetooth+Low+Energy/9781491900550/ch04.html#gatt_writes
http://stackoverflow.com/a/27492345/820942

randdusing, 2015-06-09:
Once I start working on my mobile app again, I'm going to try to update the plugin, but right now have some other priorities.

It looks like it's possible with Android as shown here: https://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#requestMtu(int)

But I don't see anything in iOS for changing the MTU unfortunately.

Tomino2112, 2015-06-09:
@randdusing  thannks for looking at it.

Actually, it is not that much about changing MTU as it cannot be changed to higher anyway, only to lower. Its more about fragmenting and defragmenting the message

ChadyG, 2015-08-19:
Changing the MTU to something larger (64) was working for me when testing on Samsung S3 devices.  I'm not sure how much interest there is, but I can prepare a simple PR with the changes.

randdusing, 2016-01-06:
@ChadyG How big of a performance improvement do you notice from bumping it up to 64? And were you primarily reading or writing?

ChadyG, 2016-01-25:
I'm not sure what the actual performance increase was, unfortunately.  We bumped it initially for better compatibility with a custom BT stack on a peripheral.  I do know it was noticably slower when we disabled the MTU change, and we are primarily doing reads with only a few attributes that go over the default size.

randdusing, 2016-01-25:
Ok, thanks for the update. I want to test changing the MTU more, but need the proper devices.

randdusing, 2016-03-31:
@Tomino2112 Try the new writeQ function with write without response. It will fragment the bytes into 20 byte fragments and write them as quickly as possible. There isn't much performance benefit on iOS, but Android throughput will increase. Also, check out Android's requestConnectionPriority which will lower the connection interval if possible.

closed 2016-03-31 by randdusing