randdusing/cordova-plugin-bluetoothle #160: Rename BLE Devices

wolke7, 2015-06-09:
Hi @randdusing,  
do you have experience with renaming BLE devices?
For example if you want to have one BLE on your car keys for tracking them. So you rename them in your App once and for every new scan it recognize the new name.

Thanks !

randdusing, 2015-06-09:
I've tried to rename one of my Polar heart rate monitors, but it didn't work. I tried using the Device Name characteristic of the Generic Access service (https://developer.bluetooth.org/gatt/services/Pages/ServiceViewer.aspx?u=org.bluetooth.service.generic_access.xml) It shows Write as Optional, but the characteristic description also states "All available Characteristics are read only".

I believe I could also "rename" the device on my Windows phone, but it reset to the default name was I unpaired the device from the Windows phone. This leads me to believe it didn't actually rename anything on the heart rate monitor.

But with that said, some devices could have their own services/characteristics that allow you to rename the device.

wolke7, 2015-06-09:
hm ok, I guess my devices don't allow to write the Characteristics, too. 

How did you "rename" the devices on your phone? 
My idea would be to safe in my App that BLE "xyz" is now called "car keys". When you tried the same and it didn't really work, I don't have to try this.

randdusing, 2015-06-09:
I don't have my Windows phone on me at the moment, so I'll have to get back to you on that.

Could you store the UUID and name in your app? Then allow the user to change the name to whatever. That way the name is changed until the device forgets about that UUID.

wolke7, 2015-06-09:
No not yet.. I will try this the next few days.

randdusing, 2016-01-24:
Hi @wolke7, still need help with this?

randdusing, 2016-03-24:
Closing due to inactivity. Please reopen if you still have any issues.

closed 2016-03-24 by randdusing

Thyrion33, 2019-05-20:
Hello,
I use your plugin for an App intended for iOS and Android.
The device is renamed via a special characteristic. 
The renaming works fine on both Android and iOS. The problem is that when I scan after renaming the device name is not changed on iOS, i have to wait some time. On Android it's work fine.
Is that some sort of cache ? Any way to fix it ?
Thanks in advance !