randdusing/cordova-plugin-bluetoothle #161: Discovery and Pairing

splinter7, 2015-06-16:
Hello,

I have currently installed your plugin in my Cordova 4.0 development app. The example file provided within the plugin works to initialize, enable and disable Bluetooth. The features I'm interested in most Discovery and pairing/connecting do not work at all. I'm testing your plugin on an Xperia M (1st gen) mobile phone running Android 4.3 with a Bluetooth 3.0 headset.

The goal I am trying to achieve is to create an app that will auto-detect the headset once Bluetooth is enabled on both the mobile device and headset and pair the two with as little user intervention as possible.

hervethouzard, 2015-07-16:
Hello,

1) It is recommended to use Android 4.4
2) A Bluetooth 3.0 Handset is not BLE compatible

Bye,
Hervé

randdusing, 2015-07-16:
Hervé is correct. This plugin only supports BLE.

splinter7, 2015-07-17:
Thanks for the reply guys. I have already solved this problem with another plugin I found shortly after posting this issue.

rafagomes, 2015-08-24:
Hey guys! Can I use this to connect two androids somehow? Thanks in advance :)

randdusing, 2015-08-24:
This plugin only supports the client side of Bluetooth LE at the moment.

rafagomes, 2015-08-24:
Thanks @randdusing . Any ideas on how share a bip (sound) on multiple phones?

randdusing, 2016-01-24:
@rafagomes With the current state of the plugin, I don't think it's possible with Bluetooth LE. Once server support is added, it's theoretically possible, but it's probably better to use something like classic Bluetooth for phone to phone data transfer.

closed 2016-01-24 by randdusing