randdusing/cordova-plugin-bluetoothle #162: "No BLE device found" with Windows 8

MeMyselfI, 2015-06-28:
The plugin can't find a device. As I'm new to Windows Store App programming: Is there anything I have to configure first or anything I can debug to find out the reason for that?

Thanks in advance.

darkriderdesign, 2016-01-27:
For starters this is a great project. -Thank you! -  I am also having this issue in Windows 10. 
BluetoothLePlugin.js

initialize: function
"No BLE devices found"

The Manifest file and its formatting was giving me some trouble but I was able work through it by borrowing from new MS documentation and creating a new Visual Studio windows 10 project and combining the two manifests.

OLD MANIFEST based on windows 8? - Capabilities bluetooth - this is currently commited
----------------------------------------------------------------------------
 <Capabilities>
        <Capability Name="internetClient" />
        <m2:DeviceCapability Name="bluetooth.genericAttributeProfile">
            <m2:Device Id="any">
                <m2:Function Type="serviceId:1800" />
                <m2:Function Type="serviceId:1801" />
                <m2:Function Type="serviceId:1802" />
                <m2:Function Type="serviceId:1803" />
                <m2:Function Type="serviceId:1804" />
                <m2:Function Type="serviceId:1806" />
                <m2:Function Type="serviceId:1807" />
                <m2:Function Type="serviceId:1808" />
                <m2:Function Type="serviceId:1809" />
                <m2:Function Type="serviceId:180A" />
                <m2:Function Type="serviceId:180D" />
                <m2:Function Type="serviceId:180E" />
                <m2:Function Type="serviceId:180F" />
                <m2:Function Type="serviceId:1810" />
                <m2:Function Type="serviceId:1811" />
                <m2:Function Type="serviceId:1812" />
                <m2:Function Type="serviceId:1813" />
                <m2:Function Type="serviceId:1814" />
                <m2:Function Type="serviceId:1816" />
                <m2:Function Type="serviceId:1818" />
                <m2:Function Type="serviceId:1819" />
                <m2:Function Type="serviceId:181A" />
                <m2:Function Type="serviceId:181B" />
                <m2:Function Type="serviceId:181C" />
                <m2:Function Type="serviceId:181D" />
                <m2:Function Type="serviceId:181E" />
                <m2:Function Type="serviceId:181F" />
                <m2:Function Type="serviceId:1820" />
                <m2:Function Type="serviceId:78b2" />
            </m2:Device>
        </m2:DeviceCapability>
        <DeviceCapability Name="microphone" />
    </Capabilities>
----------------------------------------------------------------------------

NEW WINDOWS 10 MANIFEST - Capabilities bluetooth - if you were making a new VS UWP project this works on my windows 10 phone
----------------------------------------------------------------------------
<Capabilities>
   <DeviceCapability Name="bluetooth.genericAttributeProfile">
        <Device Id="any">
            <Function Type="name:genericAccess" />
        </Device>
    </DeviceCapability>
</Capabilities>
----------------------------------------------------------------------------

COMBO WINDOWS 10 MANIFEST - Capabilities bluetooth
----------------------------------------------------------------------------

<Capabilities>
        <DeviceCapability Name="bluetooth.genericAttributeProfile">
            <Device Id="any">
                <Function Type="serviceId:1800" />
                <Function Type="serviceId:1801" />
                <Function Type="serviceId:1802" />
                <Function Type="serviceId:1803" />
                <Function Type="serviceId:1804" />
                <Function Type="serviceId:1806" />
                <Function Type="serviceId:1807" />
                <Function Type="serviceId:1808" />
                <Function Type="serviceId:1809" />
                <Function Type="serviceId:180A" />
                <Function Type="serviceId:180D" />
                <Function Type="serviceId:180E" />
                <Function Type="serviceId:180F" />
                <Function Type="serviceId:1810" />
                <Function Type="serviceId:1811" />
                <Function Type="serviceId:1812" />
                <Function Type="serviceId:1813" />
                <Function Type="serviceId:1814" />
                <Function Type="serviceId:1816" />
                <Function Type="serviceId:1818" />
                <Function Type="serviceId:1819" />
                <Function Type="serviceId:181A" />
                <Function Type="serviceId:181B" />
                <Function Type="serviceId:181C" />
                <Function Type="serviceId:181D" />
                <Function Type="serviceId:181E" />
                <Function Type="serviceId:181F" />
                <Function Type="serviceId:1820" />
                <Function Type="serviceId:78b2" />
            </Device>
        </DeviceCapability>
    </Capabilities>

randdusing, 2016-01-27:
@darkriderdesign Your post came in really funny. I'm not sure if it's missing anything.

Unfortunately I didn't write the Windows code, and not very familiar with it. I've been hoping to take a look at it, but I've been busy with my business and having a Macbook makes it difficult.

darkriderdesign, 2016-01-27:
@randdusing sorry I was missing the back dash'. so the manifests didn't show up.   Hopefully now that the manifests are there is makes more sense.  I found a windows 10 - WinJS - UWP based BLE project here.  it compiles and can find iBeacons. not sure how to convert it to Cordova though.

https://github.com/Microsoft/Windows-universal-samples/tree/master/Samples/BluetoothAdvertisement

randdusing, 2016-03-24:
@darkriderdesign Windows platform has been updated by some other contributors to update the manifest based on the Windows version. I think this should solve your issue.

closed 2016-03-24 by randdusing

darkriderdesign, 2016-03-25:
You are a sorcerer, thank you.  Let me know you ever need anything design, ui, ux related so I can return the favor

-darkriderdesign

randdusing, 2016-03-25:
@darkriderdesign  Don't thank me, thank @TimBarham. Although I did just order a Surface Pro 4 yesterday, so I should be able to start updating it in the near future as well.