randdusing/cordova-plugin-bluetoothle #164: calling startScan multiple times to get multiple devices?

gortok, 2015-07-04:
I've [implemented an Angular $q wrapper for the Bluetoothle](https://github.com/Jewelbots/ng-cordova/blob/bluetoothle-plugin-addition/src/plugins/bluetoothle.js) code; and one issue I'm running into is seeing multiple bluetoothle devices; it appears the successCallback gets called once the scanResult happens, even if there are more devices to find;  how do you handle this?

randdusing, 2015-07-04:
I'm not really familiar with Angular, but is it possible to receive multiple success responses? If not, you would have to modify the code to only return the results from the past X seconds all at once.

gortok, 2015-07-06:
In the angular $q library, a resolved promise always returns the first resolution (in this case, the first startScan result that is listed as successful.  As an aside, I [used this code](https://github.com/Jewelbots/ng-cordova/commit/88463cb6b2adf09b052c362c5928b4d9223c2f1f) to actually get a scan result:

randdusing, 2015-09-26:
I posted an example of how I implemented my scan code for angular.

https://github.com/randdusing/BluetoothLE/issues/174

closed 2015-09-26 by randdusing