randdusing/cordova-plugin-bluetoothle #165: while building project i am getting this errors or warning.

lawlessnut, 2015-07-06:
Running command: /Users/vipul/Documents/workspace/research/BLETry/platforms/android/cordova/build 
ANDROID_HOME=/Users/vipul/bin/adt-bundle-mac/sdk
JAVA_HOME=/Library/Java/JavaVirtualMachines/jdk1.7.0_65.jdk/Contents/Home
Running: /Users/vipul/Documents/workspace/research/BLETry/platforms/android/gradlew cdvBuildDebug -b /Users/vipul/Documents/workspace/research/BLETry/platforms/android/build.gradle -Dorg.gradle.daemon=true
:preBuild
:compileDebugNdk
:preDebugBuild
:checkDebugManifest
:CordovaLib:compileLint
:CordovaLib:copyDebugLint UP-TO-DATE
:CordovaLib:mergeDebugProguardFiles
:CordovaLib:preBuild
:CordovaLib:preDebugBuild
:CordovaLib:checkDebugManifest
:CordovaLib:prepareDebugDependencies
:CordovaLib:compileDebugAidl
:CordovaLib:compileDebugRenderscript
:CordovaLib:generateDebugBuildConfig
:CordovaLib:generateDebugAssets UP-TO-DATE
:CordovaLib:mergeDebugAssets
:CordovaLib:generateDebugResValues
:CordovaLib:generateDebugResources
:CordovaLib:packageDebugResources
:CordovaLib:processDebugManifest
:CordovaLib:processDebugResources
:CordovaLib:generateDebugSources
:CordovaLib:compileDebugJava
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
:CordovaLib:processDebugJavaRes UP-TO-DATE
:CordovaLib:packageDebugJar
:CordovaLib:compileDebugNdk
:CordovaLib:packageDebugJniLibs UP-TO-DATE
:CordovaLib:packageDebugLocalJar UP-TO-DATE
:CordovaLib:packageDebugRenderscript UP-TO-DATE
:CordovaLib:bundleDebug
:prepareAndroidCordovaLibUnspecifiedDebugLibrary
:prepareDebugDependencies
:compileDebugAidl
:compileDebugRenderscript
:generateDebugBuildConfig
:generateDebugAssets UP-TO-DATE
:mergeDebugAssets
:generateDebugResValues
:generateDebugResources
:mergeDebugResources
:processDebugManifest
:processDebugResources
:generateDebugSources
:compileDebugJava
Note: /Users/vipul/Documents/workspace/research/BLETry/platforms/android/src/com/randdusing/bluetoothle/BluetoothLePlugin.java uses or overrides a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/vipul/Documents/workspace/research/BLETry/platforms/android/src/com/randdusing/bluetoothle/BluetoothLePlugin.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
:preDexDebug
:dexDebug
:processDebugJavaRes UP-TO-DATE
:validateDebugSigning
:packageDebug
:zipalignDebug
:assembleDebug
:cdvBuildDebug

BUILD SUCCESSFUL

Total time: 51.56 secs
Built the following apk(s):
    /Users/vipul/Documents/workspace/research/BLETry/platforms/android/build/outputs/apk/android-debug.apk
pchm87-01M:BLETry vipul$ 

I am using cordova 5.0.0. Is it going to work ? if no please let me know which right version cordova i should use in order to use this BLE plugin.

randdusing, 2015-07-07:
Is the plugin still working? It should be working, but I will need to make some changes to get rid of those warnings.

lawlessnut, 2015-07-07:
plugin working for cordova 5.0.0 i have tried on my galaxy s6 edge.

randdusing, 2015-09-26:
Deprecation warnings will go away when new Bluetooth LE scan is added. https://github.com/randdusing/BluetoothLE/issues/191

closed 2015-09-26 by randdusing