randdusing/cordova-plugin-bluetoothle #170: Lumia 925 & 620 with Windows 8.1

hervethouzard, 2015-07-16:
Hello,

I'm using the plugin with success on iOs and Android but with two Lumias (925 & 620) under Windows Phone 8.1 I can't get the plugin to run.

All I get when I run the sample provided with the plugin is error messages like this:

Enable Error : "Missing Command Error"
Start Scan : {"serviceUuids":[]}
Start Scan Error : "Missing Command Error"

So I'd like to know if the plugin can runs on Windows Phone 8.1 ?
I'm also using Visual Studio Express 2013 for Windows	on a Windows 8.1 PC.

Bye,
Hervé

lhelsloot, 2015-07-16:
Windows Phone 8.1 does not support enable and startScan. Bluetooth devices can only be found via the phone's settings, after which they can be used from a user application via retrieveConnected.

hervethouzard, 2015-07-16:
Hello,

thanks for the information, I will try again.

bye

thomasbaggio, 2015-08-20:
Hi @hervethouzard how do you have installed the plugin for Windows Phone? 
In plugin.xml I see that the code block starting with  <platform name="wp8"> is commented... 
thanks

hervethouzard, 2015-08-21:
@thomasbaggio I did not made a Windows version...

randdusing, 2015-08-23:
I didn't develop the Windows version, so I can't really support it. I had plans to rewrite it, but those will be delayed again until at least the end of the year.

randdusing, 2016-03-24:
The Windows platform code (8.1, 10, Phone 8.1) has been updated by some other contributors. Let me know if this doesn't solve the issue.

closed 2016-03-24 by randdusing