randdusing/cordova-plugin-bluetoothle #171: IOS scan result shows previous name of BLE device

aroncal, 2015-07-16:
In IOS, if you have ever been connected to a device with name (A), and this device changes its name to (B), the next time IOS scans for bluetooth devices keeps showing (A) on the scan results.
The former name keeps showing on scan results until you connect & disconnect to the device. Then yes, (B) starts appearing on scan results instead of (A).

randdusing, 2015-07-16:
I'll take a look, but this probably an issue at the iOS stack level as I'm not cacheing any of the scan results.

aroncal, 2015-07-17:
I think you are right. I have made more tests and it seems like it's an IOS issue caching the BLE name (I'm using version 8.1.3 and this time).

closed 2015-08-23 by randdusing

ikarus-balsero, 2015-11-03:
What is the work around to this situation programmatically? Please help, thanks

aroncal, 2015-11-04:
My workaround looks something like this (when on IOS):

1. App changes BT devices's name through order. Saves new name locally.
2. Disconnect app from BT device
3. Scan & Connect app to BT device. Old name still broadcasted by the BT device. Use saved name on step 1 to fake the name while scanning and right after connecting.
5. Hopefully on next disconnect->scan->connect the new name will show up already correctly.

timfish, 2016-02-11:
Ok, this is an old issue but there does [appear to be a hook](https://developer.apple.com/library/ios/documentation/CoreBluetooth/Reference/CBPeripheralDelegate_Protocol/#//apple_ref/occ/intfm/CBPeripheralDelegate/peripheralDidUpdateName:) in iOS for when the device name changes. Could that help?

I'm guessing the device would have to support the `Service Changed` characteristic and trigger it when the name changes. My dev board here supports it and I have access to an iPad but no objective-c build system...

randdusing, 2016-02-14:
I didn't realize that hook existed. I'm wondering how I should relay those changes back to the Cordova app itself...