randdusing/cordova-plugin-bluetoothle #172: Reconnecting multiple times on Android causes auto-connect

mpedersen15, 2015-07-20:
When disconnecting and reconnecting my BLE peripheral to my Android device (Galaxy S4 with Android 4.4.4), I can successfully reconnect once. Then if I disconnect again (by turning the peripheral off) and turn my peripheral back on, the disconnect event is fired but the peripheral and phone automatically connect. If the peripheral is turned off at this point, no disconnect event is fired.

Has anyone else has this issue? I'm trying to track down other Android devices to see if this issue is universal. Any helpful hints or insights are much appreciated!

randdusing, 2015-09-26:
I haven't noticed this issue, but I usually call close after immediately after disconnecting. When the plugin calls, connectGatt its setting autoConnect to false, so I'm not sure why you're seeing this issue.

http://developer.android.com/reference/android/bluetooth/BluetoothDevice.html#connectGatt(android.content.Context, boolean, android.bluetooth.BluetoothGattCallback)

mpedersen15, 2015-09-28:
Thanks for checking this out. I'm definitely heading toward closing the connection on every disconnect to work around the issue. I haven't tested on too many Android devices, but if I'm able to and find anything interesting, I'll be sure to post.