randdusing/cordova-plugin-bluetoothle #174: bluetoothle is undefined on Ionic view with iPhone

maxrue, 2015-07-29:
Hi,
I try your plugin with the iPhone simulator and on my hardware with the ionic view app. With the simulator it says, that no bluetooth is possible. But with the hardware iPhone and the ionic view app the bluetoothle variable is undefined.
I updated all plugins, cordova and ionic, include many times the plugin and use your example program. Every time the same problem. 

Have someone the same issue with ionic view on iphone and/or could help me?

Thanks

````
document.addEventListener("deviceready", initialize, false);

function initialize () {
        alert('init');
        var params = {
            "request": true,
            "statusReceiver": true
        };

        try {
            alert(JSON.stringify(bluetoothle));
            bluetoothle.initialize(initializeSuccess, initializeError, params);
        } catch (e) {
            alert(e); // bluetoothle.initialize(...) is undefined.
        }
        function initializeSuccess(res) {
            alert(JSON.stringify(res));
        }

        function initializeError(res) {
            alert(JSON.stringify(res));
        }

        bluetoothle.isInitialized(isInitialized);

        function isInitialized(isIt) {
            alert(isIt);
        }
    }
````

randdusing, 2015-07-30:
Sorry, don't have any Ionic experience although I'm planning to convert my app to it later this year. How are you adding the plugin?

maxrue, 2015-07-30:
Hi,
thanks for your respond.
the only thing I do to add the plugin is
````
cordova plugin add https://github.com/randdusing/BluetoothLE
````
after that, cordova plugin ls list the plugin with version 2.1.0.

randdusing, 2015-07-31:
Can you see if the example app works?

randdusing, 2015-08-13:
Just an update. I've started converting my primary app over to Ionic and will then add support for the plugin as well.

maxrue, 2015-08-20:
Hi and sorry for the late response. The example app doesn't work too.

Great I'm waiting for your support for ionic. Thanks :)

Godowsky1, 2015-09-15:
any update on the ionic support?

Godowsky1, 2015-09-15:
Or maybe I can just ask you a question ;) I have a BluetoothLE device that is sending a value every 10th sec. I just want to get this information, - I work with ionic but I have no idea how to receive this information. I know nothing about the BluetoothLE device, - only that it works and that it is sending this value. I have a javaversion where it works (I haven't done that version myself) so I can see that my device actually does send the data and that an android app is able to get it, - but how to make a hybrid version that does the same, - thats the problem ;)

maxrue, 2015-09-15:
First, you must inintialize/enable bluetooth. Than scan all devices and connect yours. After that, you can discover the services and discover the characteristics. Just subscribe after all the characteristic you want. To analyse the data from the device take a look to the example from Rand in the readme.md or the example data:

https://github.com/randdusing/BluetoothLE/tree/master/example/www <https://github.com/randdusing/BluetoothLE/tree/master/example/www>

Hope it helps :)

randdusing, 2015-09-15:
@Godowsky1 I've been pretty busy working on the actual ionic version of my app. I'll be working full time in about 3 weeks, so hopefully I'll have more time to dedicate to the plugin.

sarahsga, 2015-09-16:
@randdusing In my ionic app, i copy pasted the code by @maxrue (the asker) above. Instead of getting an exception "undefined", I always get an error callback saying "Invalid action" on one of my android phones, and using same apk on another android phone, none of the callbacks is called. Please help. I think the issue isn't ionic but something else

don, 2015-09-17:
@Godowsky1 you might want to check out the angular wrapper for BluetoothLE that @gortok wrote https://github.com/Jewelbots/ng-cordova-bluetoothle

Godowsky1, 2015-09-17:
Thanks I'll do that :)

Sendt fra min iPhone

gortok, 2015-09-17:
@randdusing: feel free to fork the wrapper I've created for your plugin (or point people to it).

The wrapper definitely not bug free but it does work in our app; so there's that. I have an android phone on order so as soon as that comes, I'll recheck my code against android and see how it does.  The BLE part is still in the earlier stages; so definitely open an issue if you encounter a problem.

randdusing, 2015-09-26:
@gortok Great work. I partially implemented a wrapper for my project, but it's a bit hardcoded to my project. I'm still pretty new to Angular, so I don't know the best way to handle functions that have multiple callbacks like connect, scan, subscribe, etc.

This is how I'm currently doing it, maybe some more experienced users can let me know if it's a good or bad idea.

```javascript
var q = $q.defer();
if (window.bluetoothle === undefined) {
  q.reject();
} else {
  window.bluetoothle.startScan(
    function(obj) {
      q.resolve(obj);
      $rootScope.$broadcast('$bluetoothLE:startScan', obj);
    },
    function(obj) {
      q.reject(obj);
    },
    params
  );
}
return q.promise;
```

@maxrue Does window.bluetoothle work for you? That's how I'm accessing bluetoothle in my Ionic based app.

randdusing, 2015-09-30:
@gortok I made some changes to the angular wrapper. Mostly switched some of the functions to use the notify callback and added an Ionic-based example app. Let me know what you think and I'll add a pull request.

closed 2016-01-24 by randdusing