randdusing/cordova-plugin-bluetoothle #177: Subscription dropped on Windows

lhelsloot, 2015-08-15:
My Windows Phone 8.1 device keeps dropping its subscription to a BLE characteristic. After subscribing, it receives notifications for a short while, after which no more updates are received.

Based on [this SO question](http://stackoverflow.com/questions/28334736/gattcharacteristic-valuechanged-stops-getting-called), I explicitly prevent characteristic descriptors on which the `onvaluechanged` property is set from being garbage collected by storing them in a local object, which seems to work so far. However, I've not been able to thoroughly test this solution. Moreover, it's far from elegant.

Is anyone aware of a cleaner solution?

thomasbaggio, 2015-08-20:
@lhelsloot Hi, one question: how do have you installed the plugin for WP8? In plugin.xml i see platform wp8 commented. Does it work for Windows Phone? Thanks

dki1110, 2020-07-15:
Hello,

I faced same issue. My environment is below.

- Central: Windows10
- Peripheral: Pulse Oximeter (A indication occur when finger is pulled out from device.)

`subscribe` event occured (writeClientCharacteristicConfigurationDescriptorAsync is succeed).
But, when finger is pulled out from device, `subscribeResult` event did not occur.
I tried it several times and `subscribeResult` event happened only once. The timing of `subscribe` event and pulling out finger at that time was almost the same.

Is there any good way?

Thanks