randdusing/cordova-plugin-bluetoothle #180: Connecting to RN4020, MLDP structure

Janckk, 2015-08-19:
Hi guys,

I have this BLE microchip I'm trying to write to. The thing is that I can connect to it, discover it, read from it but I'm not sure how to send the proper write data to it.

I'm trying to send this simple data "1\n" -> "0x31 0x0A" (hex). I know it uses some type of mldp structure but there is a lack of documentation on their side, I've also contacted support and waiting for response).

Has anyone had any experience with this?

Janckk, 2015-08-24:
I've attached the screenshots and more wireshark debugging info here:

http://stackoverflow.com/questions/32181100/android-low-energy-bluetooth-connection-sending-data-wireshark-debugging

randdusing, 2015-09-26:
@Janckk I've updated the documentation with a little better example of writing. You'll need to convert the HEX code to bytes, then the bytes to an encodedString and set that as the value. Doesn't look like your stackoverflow link works anymore.

closed 2016-01-24 by randdusing