randdusing/cordova-plugin-bluetoothle #185: Android 5.0: connect() extremely unreliable but works fine on Android 4.4.x

mrtomhoward, 2015-08-25:
There seems to be a known issue with connect reliability to BLE peripherals with Android 5.0 and later.

There is a discussion about it on Google Code, but with no acknowledgement from the Android team: https://code.google.com/p/android/issues/detail?id=82293

But I couldn't find any mention of it in this plugin's issues, so I wondered if there is some fix or workaround I'm missing?

I discovered this issue when I upgraded my Samsung Galaxy S5 from 4.4.x (KitKat) to Lollipop.

On KitKat, my app using the BluetoothLE plugin was working perfectly.  But from the instant I upgraded to Lollipop, reliability dropped through the floor, and it often takes 15 or more connect attempts, each with a timeout of up to 15 seconds, to establish a connection.

The app detects advertising frames via StartScan no problems, but when connect() is initiated, nothing happens, even with a timeout of 20-30 seconds (the peripheral's advertising rate is 1/5s to minimise power consumption).

The BLE hardware device is the Laird BL600.

Everything still works perfectly on KitKat and iOS.

Any ideas?

Thanks,
Tom

akiwarheit, 2015-08-28:
It's more of a Lollipop issue if you ask me. My issue is more native Android's bluetooth API; `connectGatt` immediately going to `STATE_DISCONNECTED` not even passing `STATE_CONNECTED` nor `STATE_CONNECTING`

Pre-lollipop, it works flawlessly.

mrtomhoward, 2015-08-29:
Yeah, all the research and testing I've done seems to indicate it's a Lollipop issue with Bluetooth generally.

These discussions all refer to it:
https://code.google.com/p/android/issues/detail?id=79488
https://code.google.com/p/android/issues/detail?id=82293
https://code.google.com/p/android/issues/detail?id=156730

According to one of these discussions, it's slated to be fixed in Marshmallow.  It's quite astonishing it has been allowed to remain unresolved for so long.

I find that killing the app and clearing data/cache in Bluetooth from App Manager fixes it for one connection session, but then the problem returns immediately for subsequent connections.  

But with our app, normal use of the system requires multiple connects/disconnects, sometimes several times per day, so requiring users to do this is far from satisfactory.

Oh well, just another reason to favour iOS I guess.

paultanner, 2015-09-24:
Thx for this. I thought I it was my code but I am indeed on droid 5.01 so maybe not.  While investigating I wondered whether I was calling reconnect or disconnect wrongly.  Now I notice that those two apis may need to have another argument in the call that is not clearly shown in the api document, specifically the params one that provides the address of the device to disconnect from or reconnect to.  It might be an idea to fix the doc if so.

If these were wrongly called we could fail to reconnect and also maybe hang onto the connection when exiting, thereby making it had to connect when the app is restarted?  (I'm using this with cordova.)

Paul

mrtomhoward, 2015-09-24:
From my most recent testing, the problem
does seem to be much worse from my Cordova app using BluetoothLe, compared
with the Laird toolkit app (which uses native code of course).

The Laird app is still somewhat flakey but does end up connecting within
2-3 attempts after a total of about 10-15 secs.

Whereas my Cordova/BluetoothLe app takes 15+ connect attempts over 3+
minutes on Android 5, but still works very fast and reliably on 4.4.2.

randdusing, 2015-09-26:
When you guys try reconnecting are you immediately trying to reconnect or setting a small timeout? Some previous connection issues were helped with a short timeout.

paultanner, 2015-09-27:
Yes, I have added that. All working ok on droid 4.4.4.  Back to testing on 5.01 now.

mrtomhoward, 2015-10-04:
Something occurred to me that may give some insight into the issue:

When my timeout handler executes on the connect() call, I call
disconnect(), and find that the Success function executes pretty much every
time, even though the connection status hasn't fired 'connected'.

So it seems that the connection is stuck in some in-between state; not
completing a connection successfully enough to fire 'connected' to the
connect() call, but enough to call the success function when disconnect is
called.

Hope this helps.

mrtomhoward, 2015-10-16:
OK, I now have it working quite reliably on Android 5.x, by doing the following:

 - Call bluetoothle.isConnected().  
 - If isConnected == false, run bluetoothle.disconnect().
 - Repeat bluetoothle.disconnect() call until connection is definitely disconnected (I.e., repeat disconnect() at 1000ms interval until disconnectError executes. 
 - In disconnectError, call bluetoothle.close().
 - Repeat bluetoothle.close() every 1000ms until closeError executes.
 - In closeError, set a timeOut of 5000ms, then call bluetoothle.connect().
 - When finished transaction, call disconnect() and close().

When I do this, it seems to work first time every time.  

I'll keep trying to optimise by seeing how low I can get the timeouts while keeping it reliable, but the main point is that this process seems to succeed in getting it to work reliably.

On Android 4.x and iOS, I still use bluetoothle.reconnect() if isConnected == false, and it works fine.

diegosoek, 2016-05-29:
Hello, you solved this problem? If so how did?

mrtomhoward, 2016-06-01:
In my most recent work using this plugin, I've been using a more
lightweight framework - Mobile Angular UI, rather than Sencha Touch, which
I used on the project on which this issue initially arose.

In the more recent project (using Mobile Angular UI) connectivity seems
dramatically more reliable.

It may be that the issue is strongly influenced by CPU load and/or memory
usage.

randdusing, 2016-06-01:
That's kind of scary. I generally recommend just calling close() on any sort of disconnect and then calling connect() if you'd like to reconnect.

mrtomhoward, 2016-06-02:
> I generally recommend just calling close() on any sort of disconnect and then calling connect() if you'd like to reconnect.

I do that too.  I always wait for confirmation that close() has completed before re-attempting connect().  But there still seems to be a big difference in reliability between Sencha Touch and Mobile Angular UI.

wangxiao, 2016-08-07:
I have the same question, connect the device need a long time whatever success or failed, and may be retry 20 times, connect success.

Android 5.1

my code like this:
```javascript
var connectTimes = 0;
var scan = function() {
  bluetoothle.startScan(function(res) {
    if (res.address && !connectTimes) {
      connect(res.address);
      stopScan();
      connectTimes ++;
    }
  });
};

function stopScan() {
  bluetoothle.stopScan();
}

function connect(address) {
    bluetoothle.connect(function(res) {
        alert('ok');
    }, function(err) {
        bluetoothle.close(function(){
             connect(address);
        }, function(){}, {
            address: address,
        });
    }, {
      address: address,
    });
}
```

hazemhagrass, 2016-10-29:
not working with me too! any ideas!

randdusing, 2016-10-31:
Hard to fix without being able to fix locally on a device.. and it seems like an Android issue.

But @mrtomhoward listed some steps that helped him.