randdusing/cordova-plugin-bluetoothle #188: Android build issue

basvasilich, 2015-09-21:
Hi. I've experienced some issues with android build process

```
:compileDebugJava
/Users/bs/Projects/app/platforms/android/src/com/randdusing/bluetoothle/BluetoothLePlugin.java:501: error: cannot find symbol
    else if (mtuActionName.equals(action))
             ^
  symbol:   variable mtuActionName
  location: class BluetoothLePlugin
/Users/bs/Projects/app/platforms/android/src/com/randdusing/bluetoothle/BluetoothLePlugin.java:2727: error: incompatible types: int cannot be converted to String
    String mtu = obj.optInt(keyMtu);
                           ^
/Users/bs/Projects/app/platforms/android/src/com/randdusing/bluetoothle/BluetoothLePlugin.java:2729: error: incomparable types: String and int
    if (mtu == 0)
            ^
/Users/bs/Projects/app/platforms/android/src/com/randdusing/bluetoothle/BluetoothLePlugin.java:2734: error: incompatible types: String cannot be converted to int
    return mtu;
```

My target SDK:
```
    <uses-sdk android:minSdkVersion="19" android:targetSdkVersion="23" />
```

randdusing, 2015-09-21:
Are you sure you're using the latest version? I haven't added MTU yet. There's a pull request with it, but it requires a few changes before merging. I merged it into my code, but haven't been able to test it much yet.

basvasilich, 2015-09-21:
I've run `cordova plugin add com.randdusing.bluetoothle`

As a result I have
```
  "name": "com.randdusing.bluetoothle",
  "version": "2.1.0",
```
 in my plugins

randdusing, 2015-09-21:
Oops, I was confused on how NPM works. For some reason I thought it was pulling in code from GitHub rather than the working version of my code. I've fixed this. Just remove and readd the plugin.

closed 2015-09-21 by randdusing

basvasilich, 2015-09-21:
Yes, everything looks good, thanks.