randdusing/cordova-plugin-bluetoothle #189: Support for Windows 8.1 and 10

hanthomas, 2015-09-22:
Are there any plans to make this plugin work on Windows 8.1 / 10 platform?  Is there a way to make the current version work in Windows 8.1 and 10?  Thanks.

hanthomas, 2015-09-23:
It looks like Windows 8.1/10 is supported by this plugin.  It works very differently from Android and iOS in that there's no support for scanning or even to connect to a device.  But in some ways it's a whole lot simpler.  I was able to get a list of paired devices via retrieveConnected and successfully subscribe.  However, I am not getting any subscribedResults events firing.  When I examined the plugin code, it appears that characteristic.onvaluechanged never fires.  The same operation works fine in Android and iOS,  Can someone provide a solution or workaround for this problem?  Thanks.

hanthomas, 2015-09-23:
This issues can be resolved once PR #190 has been merged.

randdusing, 2015-09-23:
Merged your PR!

closed 2015-09-23 by randdusing