randdusing/cordova-plugin-bluetoothle #192: bluetoothle.disconnect documentation issue?

paultanner, 2015-09-27:
I am trying to find the cause of connect issues on android 5.01
Connection works fine if the app was stopped.  But not if it has been running in the background.
So I'm thinking that the problem is that it never properly disconnects and that causes subsequent  connect failures.

My code:
```
   if (!bluetoothle.isConnected({address:address})) next();	
  _bleLog("trying disconnection from: " + address + " with no timeout");
  var paramsObj = {
    "address": address
  };
  bluetoothle.disconnect(
    function (obj) {
	    if (obj.status == "disconnected") {
	     _bleLog("disconnected from : " + obj.name + " - " + obj.address);
	    	ble_connected=false;
		next();
	    } else if (obj.status == "disconnecting") {
	      _bleLog("disconnecting from : " + obj.name + " - " + obj.address);
	    } else {
	      _bleLog("ignoring unexpected disconnect status: " + obj.status);
	    }
    },
    function (error) {
      console.log('disconnect failed '+error.message);
      next();
    },
    paramsObj
);		  
```
is failing with this error 
```
D/CordovaLog(17392): file:///android_asset/www/cordova.js: Line 1058 : processMessage failed: Error: TypeError: callback.success.apply is not a function
```
I have tried with and without the third parameter.  The documentation is ambiguous as it states that a param object is involved but not where to place it in the call.  I have tried it as first and third parameter. 

BTW. This happens on both droid versions I'm testing with (4.4.4 and 5.05).

Paul

randdusing, 2015-09-28:
ParamsObj is in the right place. I've made a note to update the documentation.

On the issue itself, I've never seen an error like that. It looks like Cordova is having trouble calling the success function.

paultanner, 2015-09-28:
Ok thx.  I have now investigated further and this seems to be an unintended consequence of the first line of my code.  Ironically, the error caused by that first line:
```
D/CordovaLog(13592): file:///android_asset/www/cordova.js: Line 1060 : processMessage failed: Message: S01 BluetoothLePlugin1196019499 {"isConnected":false}
```
... *follows* the error in my earlier post.

If I remove the call to bluetoothle.isConnected I get no errors from bluetoothle.disconnect.  However, in this case, despite getting the expected callbacks, the disconnection either does not happen or the connection state becomes invalid in some way.  This is non-critical on droid 4.4.4 but on 5.01 usually has the effect of preventing further connections.

I am wondering if there's a workaround that would stop the app, as opposed to putting it in the background.  I know this is not correct android practice and hard to do from cordova anyway.  So for the moment I'll do more testing without the calls to isConnected and see how often reconnections fail.

Paul

randdusing, 2016-01-24:
Oops, I just realized your isConnected call didn't have the success or error callbacks. Let me know if this is still an issue after adding the callbacks.

closed 2016-01-24 by randdusing

paultanner, 2016-01-30:
Thx @Rand.  The callbacks seem to be working fine on version 3.0.1 Best/ 
Paul