randdusing/cordova-plugin-bluetoothle #199: Android 4.4.2: scanning works and then doesn't?

jondavidford, 2015-10-29:
This is probably some type of user error on my part, but posting here in case any one else has seen something like this. My bluetoothle.startScan() function returned found devices exactly once, then never worked again. Now it only will call the success callback once, with {"status": "scanStarted"}. The error callback is never called. Has anyone seen this before?

randdusing, 2015-10-29:
If you post the related callback functions, I can take a look. I'd double
check for any runtime errors and make sure you can see the Bluetooth LE
devices within the Android Bluetooth settings page as well

jondavidford, 2015-10-29:
Yes, I can see the bluetooth LE device in my android settings page (my macbook pro with bluetooth LE supported). The callback functions are the ones from your example.

```
$scope.startScan = function() {
      //TODO Disconnect / Close all addresses and empty

      var paramsObj = {serviceUuids:[], allowDuplicates: true};

      logger("Start Scan : " + JSON.stringify(paramsObj));

      bluetoothle.startScan(startScanSuccess, startScanError, paramsObj);

      return false;
    }

    function startScanSuccess(obj) {
      logger("Start Scan Success : " + JSON.stringify(obj));

      if (obj.status == "scanResult")
      {
        logger("Scan Result");
      }
      else if (obj.status == "scanStarted")
      {
        logger("Scan Started");
      }
      else
      {
        logger("Unexpected Start Scan Status");
      }
    }

    function startScanError(obj) {
      logger("Start Scan Error : " + JSON.stringify(obj));
    }
```

randdusing, 2015-10-29:
You Mac supports Bluetooth LE, but most likely isn't actively advertising itself unless you're running another app to do so. Without advertising itself, the Bluetooth LE scan won't find anything. Also, you should check out the Angular wrapper in the readme!

randdusing, 2015-12-23:
Still having any issues?

closed 2015-12-23 by randdusing

jondavidford, 2015-12-23:
I actually haven't gotten around to looking at this again, but I'll let you know if I see any issues in the future.