randdusing/cordova-plugin-bluetoothle #200: NPM Versioning

diverJohn, 2015-10-29:
Rand,

I am using Phonegap Build and wanted to try your new NPM version 2.4.0 but I get the following build error:
Error - Plugin (or a specific plugin version) not found in external repo: com.randdusing.bluetoothle@2.4.0": Fetching plugin "com.randdusing.bluetoothle@2.4.0" via npm Failed to fetch plugin com.randdusing.bluetoothle

diverJohn, 2015-10-29:
Continued...

From my config.xml file:

Version 2.4.0 fails...
<gap:plugin name="com.randdusing.bluetoothle" source="npm" version="2.4.0" />    

I was running version 2.1.2 which builds successfully.
<gap:plugin name="com.randdusing.bluetoothle" source="npm" version="2.1.2" />   

Can you also add a call that returns the plugin version so I can print the plugin version to my log file?

Thanks,
John

diverJohn, 2015-10-29:
....
    <gap:plugin name="com.randdusing.bluetoothle" source="npm" version="2.1.2" />  
    <gap:plugin name="com.randdusing.bluetoothle" source="npm" version="2.4.0" />  
....

diverJohn, 2015-10-29:
Code snippet keeps getting deleted...
    gap:plugin name="com.randdusing.bluetoothle" source="npm" version="2.1.2"
    gap:plugin name="com.randdusing.bluetoothle" source="npm" version="2.4.0"

randdusing, 2015-10-29:
Check out the updated readme, the plugin name has been changed.

<gap:plugin name="cordova-plugin-bluetoothle" source="npm" />

closed 2015-10-29 by randdusing