randdusing/cordova-plugin-bluetoothle #202: advertise / peripheral role?

apptum, 2015-11-06:
Hey randdusing, thanks for your work on this!
In your todos there's : "Support for Peripheral role on iOS 6.0+ and Android 5.0+"
You are not by any chance already working on this?

randdusing, 2015-11-10:
Unfortunately haven't begun and don't plan to in the near future. Just too busy with my primary project that doesn't require the peripheral role.

tfmontague, 2015-11-17:
Need this feature as well. Need to connect two phones over bluetooth and there's not any other PhoneGap bluetooth plugin that supports peripheral mode. I'll start a fork and begin development on these features.

- Tim

apptum, 2015-11-21:
hey tfmontague, Peter Metz already has it working and everything is fine (exept advertise as beacon in background), but his project seems to be on hold unfortunatly: https://github.com/petermetz/cordova-plugin-ibeacon

tfmontague, 2015-11-21:
@apptum - thanks for the code.

Apple also documents code for connecting two iPhones.
https://developer.apple.com/library/ios/samplecode/BTLE_Transfer/Introduction/Intro.html

p1nox, 2016-01-30:
First of all, awesome work on this plugin man @randdusing!

Do you have a possible date when you're starting this? And if not, what would be the roadmap to support this feature?

I've never build a cordova plugin, but maybe I can help. I would like to use this for a communication protocol between phones I've been working on, and of course, this is needed for the transport layer of the system.

randdusing, 2016-01-30:
@p1nox I've been working on it recently, but still needs a bit of work. Hopefully I can create a new branch within a week or two and let y'all start using it.

p1nox, 2016-01-31:
Awesome! Thanks @randdusing :+1:

randdusing, 2016-03-09:
I've added server support for iOS in [4.0.0-dev branch](https://github.com/randdusing/cordova-plugin-bluetoothle/tree/4.0.0-dev). It would be nice to have some other users test it out.

Warnings:
- I don't recommend using this in production.
- Android is completely untested.
- Method names / overall process is subject to change, especially once I align iOS and Android.
- Documentation isn't very good right now. Check out the Angular wrapper's example in the [4.0.0-dev branch](https://github.com/randdusing/ng-cordova-bluetoothle/tree/4.0.0-dev).

randdusing, 2016-03-24:
Check out the beta release: https://github.com/randdusing/cordova-plugin-bluetoothle/tree/v4.0.0-beta.1

Android has been tested and the documentation is much better. Please open a new issue if you run into any bugs.

closed 2016-03-24 by randdusing

p1nox, 2016-03-24:
Awesome work @randdusing :+1: 

I'll test it (probably this weekend), and if I find anything I wont hesitate to open an issue :feelsgood:.