randdusing/cordova-plugin-bluetoothle #203: How to install or run this app? "Current working directory is not a Cordova-based project."

tomachinz, 2015-11-07:
Hi,
I'm having trouble running this.

Ive cloned it to my Mac desktop using github to here:
cd /Users/tom/GitHub/ng-cordova-bluetoothle/

But when I do this:
cordova plugin add cordova-plugin-bluetoothle
Current working directory is not a Cordova-based project.

randdusing, 2015-11-07:
You need to set up your own Cordova project first and then copy in the files from example

cvuser0, 2015-11-24:
ng-cordova-bluetoothle is **not** a cordova project, the files you need for using the example are in the ng-cordova-bluetoothle/example folder.

You need to create a new cordova project somewhere and drag & drop the files in the example folder into the www folder in the cordova project root.

closed 2015-11-24 by randdusing

tomachinz, 2015-11-25:
Thanks that works very helpful!