randdusing/cordova-plugin-bluetoothle #204: Scan in Android 6.0

randdusing, 2015-11-10:
There are some permission changes with Android 6.0 that seem to be causing scans to not work. I'm working on an update that should hopefully fix the issue.

randdusing, 2015-11-10:
After doing some research, scanning for Bluetooth device's on Android 6.0 seems to require the Coarse Location permission. I modified the code to request for permission, but it's a little 'scary' looking to the average user since it asks "Allow {{appName}} to access this device's location?" If the user clicks deny, the error callback will return with an error named 'permissions'. You can use this to explain to the user why you're requesting permissions. I have a feeling this prompt will increase the number of support tickets I receive and decrease my ratings. :(

By the way, if the user has paired the device and you use retrieveConnected, the coarse location permissions aren't needed.

I also checked out the BluetoothLeScanner class introduced in API 21, but it requires coarse location permissions as well. Since API 21 and 22 don't request permissions at runtime, I'd need to add the Coarse Location permission to plugin.xml. I'll probably hold off on implementing this until I better understand the permissions impact.

agoransson, 2015-11-10:
I've tried adding the permissions (BT, BTAdmin, Coarse) to both Manifest and plugin.xml, but it doesn't have any effect.

randdusing, 2015-11-10:
Yah, even if you specify the permissions you still need to request to actually use them. Give me a few minutes and I'll have a dev branch with the changes.

randdusing, 2015-11-10:
https://github.com/randdusing/BluetoothLE/tree/2.5.0

randdusing, 2015-11-10:
@agoransson You'll need to update to the latest Android Cordova version 5.0.0. I'm seeing if I can enforce that via the plugin.xml

agoransson, 2015-11-10:
I found some more information now.

```
11-10 20:37:57.487  3604  3604 I chromium: [INFO:CONSOLE(177)] "Is Initialized", source: file:///android_asset/www/js/controllers.js (177)
11-10 20:37:57.590  3604  3604 I chromium: [INFO:CONSOLE(13)] "Uncaught TypeError: Cannot read property 'Keyboard' of undefined", source: file:///android_asset/www/js/app.js (13)
11-10 20:37:57.613  3604  3604 I chromium: [INFO:CONSOLE(183)] "Is Initialized Success : {"isInitialized":false}", source: file:///android_asset/www/js/controllers.js (183)
11-10 20:37:57.613  3604  3604 I chromium: [INFO:CONSOLE(191)] "Is Initialized : false", source: file:///android_asset/www/js/controllers.js (191)
11-10 20:37:57.653  3604  3604 I chromium: [INFO:CONSOLE(177)] "Is Initialized", source: file:///android_asset/www/js/controllers.js (177)
11-10 20:37:57.711  3604  3604 I chromium: [INFO:CONSOLE(183)] "Is Initialized Success : {"isInitialized":false}", source: file:///android_asset/www/js/controllers.js (183)
11-10 20:37:57.712  3604  3604 I chromium: [INFO:CONSOLE(191)] "Is Initialized : false", source: file:///android_asset/www/js/controllers.js (191)
11-10 20:37:58.751  3604  3604 I chromium: [INFO:CONSOLE(0)] "'Touch.webkitRadiusX' is deprecated and will be removed in M47, around November 2015. Please use 'Touch.radiusX' instead.", source:  (0)
11-10 20:37:58.752  3604  3604 I chromium: [INFO:CONSOLE(0)] "'Touch.webkitRadiusY' is deprecated and will be removed in M47, around November 2015. Please use 'Touch.radiusY' instead.", source:  (0)
11-10 20:37:58.753  3604  3604 I chromium: [INFO:CONSOLE(0)] "'Touch.webkitRotationAngle' is deprecated and will be removed in M47, around November 2015. Please use 'Touch.rotationAngle' instead.", source:  (0)
11-10 20:37:58.753  3604  3604 I chromium: [INFO:CONSOLE(0)] "'Touch.webkitForce' is deprecated and will be removed in M47, around November 2015. Please use 'Touch.force' instead.", source:  (0)
11-10 20:37:58.770  3604  3604 I chromium: [INFO:CONSOLE(33)] "Initialize : {"request":true}", source: file:///android_asset/www/js/controllers.js (33)
11-10 20:37:58.799  3604  3604 I chromium: [INFO:CONSOLE(39)] "Initialize Success : {"status":"enabled"}", source: file:///android_asset/www/js/controllers.js (39)
11-10 20:37:58.799  3604  3604 I chromium: [INFO:CONSOLE(43)] "Enabled", source: file:///android_asset/www/js/controllers.js (43)
11-10 20:37:58.817   799  1507 I ActivityManager: Killing 2810:se.fskab.android.reseplaneraren/u0a95 (adj 15): empty #17
11-10 20:38:04.376  3604  3604 I chromium: [INFO:CONSOLE(104)] "Start Scan : {"serviceUuids":[],"allowDuplicates":false}", source: file:///android_asset/www/js/controllers.js (104)
11-10 20:38:04.383  3604  3686 D BluetoothAdapter: startLeScan(): null
11-10 20:38:04.389  3604  3686 D BluetoothAdapter: STATE_ON
11-10 20:38:04.397   392   417 D BtGatt.GattService: registerClient() - UUID=0fb3e5f1-9978-401a-831b-95a2d310662c
11-10 20:38:04.413   392   440 D BtGatt.GattService: onClientRegistered() - UUID=0fb3e5f1-9978-401a-831b-95a2d310662c, clientIf=5
11-10 20:38:04.413  3604  3616 D BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5
11-10 20:38:04.413   392   547 D BtGatt.GattService: start scan with filters
11-10 20:38:04.415   392   445 D BtGatt.ScanManager: handling starting scan
11-10 20:38:04.416   392   445 D BtGatt.ScanManager: configureRegularScanParams() - queue=1
11-10 20:38:04.416   392   445 D BtGatt.ScanManager: configureRegularScanParams() - ScanSetting Scan mode=2 mLastConfiguredScanSetting=-2147483648
11-10 20:38:04.416   392   445 D BtGatt.ScanManager: configureRegularScanParams - scanInterval = 8000configureRegularScanParams - scanWindow = 8000
11-10 20:38:04.416   392   440 D BtGatt.GattService: onScanParamSetupCompleted : 0
11-10 20:38:04.418  3604  3604 I chromium: [INFO:CONSOLE(110)] "Start Scan Success : {"status":"scanStarted"}", source: file:///android_asset/www/js/controllers.js (110)
11-10 20:38:04.420  3604  3604 I chromium: [INFO:CONSOLE(120)] "Scan Started", source: file:///android_asset/www/js/controllers.js (120)
11-10 20:38:08.301   392   440 D bt_btif_gattc: btif_gattc_update_properties BLE device name=SensorTag len=9 dev_type=2
```

I'm particularly interested in `11-10 20:38:04.383  3604  3686 D BluetoothAdapter: startLeScan(): null`

randdusing, 2015-11-10:
It shouldn't be an issue. I believe it just means no service UUIDs are used for filtering.

Are you getting a permissions prompt at all? Since it doesn't seem to be calling the onLeScan callback, it must be getting stopped at a lower level somewhere.

What's your javascript code look like?

agoransson, 2015-11-10:
No permission prompt at all.

I'm using https://github.com/randdusing/ng-cordova-bluetoothle - the only modification is multiple controller and template files.

randdusing, 2015-11-10:
Are you sure you're using recently published code?

agoransson, 2015-11-10:
Yep, installed with `ionic plugin add https://github.com/randdusing/BluetoothLE.git#2.5.0` seems to be the correct version installed.

randdusing, 2015-11-10:
Ahh, can you check the app settings? Maybe the location permission is already set to yes. But that leads us back to the original problem.

agoransson, 2015-11-10:
Hmm. Found it.

Had to enable "Location" in the device settings.

Getting all sorts of of scan results now.

randdusing, 2015-11-10:
Interesting that it never prompted you for the permission, that's what it should have done... Glad it's working either way though!

elafargue, 2015-11-17:
My 2 cents: many apps use BTLE to actually acquire coarse location by relying on known bluetooth Beacons... so in a way, this makes sense from a user information standpoint. Not very fair to apps that use BTLE for more worthy usage...

randdusing, 2015-11-17:
Yep, at first I was annoyed, but I understand the permission requirement.

closed 2015-12-23 by randdusing