randdusing/cordova-plugin-bluetoothle #205: After Connect Success I get connecting again

doughri, 2015-11-13:
I'm using the plugin on android 6 with your example when I try to connect to BT device I get connect success then the application try to connect again, But when I try to write something I get error isNotConnected I tried on android 4.4.2 I get the same issue,
Any Ideas?

traces:
Connect : {"address":"XX:XX:XX:XX:XX:XX"}
index.js (1144,11)
Connect Success : {"status":"connecting","address":"XX:XX:XX:XX:XX:XX","name":"Device_name"}
index.js (1144,11)
Connecting

Write : {"address":"XX:XX:XX:XX:XX:XX","value":"0200130301","serviceUuid":"180D","characteristicUuid":"2A19"}
index.js (1144,11)
Write Error : {"error":"isNotConnected","message":"Device isn't connected","address":"XX:XX:XX:XX:XX:XX","name":"Device_name"}
index.js (1144,11)

randdusing, 2015-11-13:
It doesn't look like you are connected, but stuck on connecting.

doughri, 2015-11-13:
That's right apparently the target device doesn't support Bluetooth Low energy

closed 2015-12-23 by randdusing