randdusing/cordova-plugin-bluetoothle #206: iOS : window.bluetoothle undefined

jandureu, 2015-11-13:
Hello,

I'm creating an app with ionic and Visual studio 2015, with your BLE plugin. The problem is that i have this error when debugging on my device ( iPhone 6s) : window.bluetoothle undefined . 
So I can't initialize or scan anything.

I'm using the ngCordovaBluetoothLE to acces the plugin functions.

The same ionic project on a windowsPhone8.11 (Lumia 640) is OK. I can initialize, scan and find a BLE Device.

Do you have any idea ?

Thank you 

Jerome

randdusing, 2015-11-13:
Can you post any code? I'm using Ionic as well

jandureu, 2015-11-16:
Hello,

Here is the code : 

In the index.html:
```
 <!-- ionic/angularjs js -->
    <script src="scripts/platformOverrides.js"></script>
    <script src="js/framework/ionic.bundle.min.js"></script>
    <script src="js/framework/ngCordovaBluetoothLE.js"></script>
    <script src="cordova.js"></script>
    <!-- your app's js -->
    <script src="js/app.js"></script>
```

in the app.js

```
var app = angular.module('roomMngt', ['ionic', 'controllers', 'BLEcontroller', 'Ble.services', 'ngCordovaBluetoothLE']);

app.run(function ($ionicPlatform) {
    $ionicPlatform.ready(function () {
        // Hide the accessory bar by default (remove this to show the accessory bar above the keyboard
        // for form inputs)
        if (window.cordova && window.cordova.plugins && window.cordova.plugins.Keyboard) {
            cordova.plugins.Keyboard.hideKeyboardAccessoryBar(true);
        }
       
        if (window.StatusBar) {
            // org.apache.cordova.statusbar required
            StatusBar.styleLightContent();
        }
        if (window.bluetoothle) {
         
        }
        else {
            alert('no bluetooth plugin');
        }
    });
})
```

in bleSelection.html

```
<section class="home-container">
    <div ng-controller="HomeCtrl">
        <div class="row text-center">
            <div class="col"><*a class="button button-balanced button-small" ng-click="initialize()">Intialize</a></div>
......
```

in the BLEController.js

```
var app = angular.module('BLEcontroller', [])

app.controller('HomeCtrl', function ($scope, $rootScope, $state, $cordovaBluetoothLE) {
    $rootScope.devices = {};

    $scope.goToDevice = function (device) {
        $state.go("device", { address: device.address });
    };

    $scope.isEmpty = function () {
        if (Object.keys($rootScope.devices).length === 0) {
            return true;
        }
        return false;
    };

    $rootScope.initialize = function () {
        var params = { request: true };

        console.log("Initialize : " + JSON.stringify(params));

        $cordovaBluetoothLE.initialize(params).then(null, initializeError, initializeSuccess);
    };

    function initializeSuccess(obj) {

        console.log("Initialize Success : " + JSON.stringify(obj));

        if (obj.status == "enabled") {
            console.log("Enabled");
        }
        else {
            console.log("Unexpected Initialize Status");
        }
    }

    function initializeError(obj) {

        console.log("Initialize Error : " + JSON.stringify(obj));
    }
```

I always go to the function initializeError(obj) {}

Any idea ?

jandureu, 2015-11-16:
Hello,

After some debugging from a macBook with Safari attached to the device, it seems that the file "bluetoothle.js" from the www plugin folder is not included. 
I can't see this file in the resources tab, in the "all Resources" selection.

No idea why this file is not present.

Thank you

randdusing, 2015-11-16:
May want to try removing the plugin and adding it back.

randdusing, 2015-11-16:
Oops, only saw the first part in the email. Glad it's working.

closed 2015-12-23 by randdusing