randdusing/cordova-plugin-bluetoothle #207: Not scanning unpaired device

irfan8812, 2015-11-16:
Hello,

When we click on Start Scan then Retrieve Connected it only shows the paired Bluetooth IDs. Please give some suggestion how it will show all pair and unpair Bluetooth IDs.

randdusing, 2015-11-16:
What Android version are you on?

irfan8812, 2015-11-16:
We are using android 5.0.2

randdusing, 2015-11-16:
Do you see anything in the logs about scan results?

irfan8812, 2015-11-16:
The value of obj.status is 'scanStarted' everytime. It doesn't go inside
if(obj. status == "scanResult")  block. So it's not add any new device
through addDevice().

randdusing, 2015-11-16:
I mean from LogCat, then you should see if the results are even appearing at a lower level.

irfan8812, 2015-11-17:
[image: Inline image 1]

Regards,
Mohammad Irfan
Software Engineer , Interbiz Consulting
Phone: (+91-7752) 401-858, 261-189; (+1-602) 218-4062
Web: www.interbizconsulting.com ‘success through synergy’

randdusing, 2015-11-17:
Your image didn't come through although text input would be easier to copy and paste if needed.

randdusing, 2015-12-23:
Reopen if you still need help

closed 2015-12-23 by randdusing