randdusing/cordova-plugin-bluetoothle #208: App hard crash on Nexus 6P / Android 6.0

elafargue, 2015-11-17:
I have experienced quite a few hard crashes ("unfortunately, the application has stopped" message) on the Nexus 6P whenever a bluetooth device that is connected gets out of range. Same code seems to work perfectly well on Nexus 5 on Android 6.0. Will do more tests and report.

Is this a known issue on Android 6.0 ?

randdusing, 2015-11-17:
Have the crash log?

elafargue, 2015-11-17:
in a typical manner, I can't reproduce when connected to the debugger... I'll keep trying

elafargue, 2015-11-18:
OK, looks like it's related to the graphics driver... happens at random. Nexus 6P is the only phone that suffers from it as far as I can tell, not very good news:

```
1-17 17:00:18.590  25264-25321/? W/google-breakpad﹕ ### ### ### ### ### ### ### ### ### ### ### ### ###
11-17 17:00:18.590  25264-25321/? W/google-breakpad﹕ Chrome build fingerprint:
11-17 17:00:18.590  25264-25321/? W/google-breakpad﹕ a0.1.7
11-17 17:00:18.590  25264-25321/? W/google-breakpad﹕ 1070
11-17 17:00:18.590  25264-25321/? W/google-breakpad﹕ 61966e6e-3fdc-4984-9d79-401e23ee5095
11-17 17:00:18.590  25264-25321/? W/google-breakpad﹕ ### ### ### ### ### ### ### ### ### ### ### ### ###
11-17 17:00:18.590  25264-25321/? A/libc﹕ Fatal signal 11 (SIGSEGV), code 1, fault addr 0x440 in tid 25321 (GpuThread)
11-17 17:00:18.693    3829-3829/? A/DEBUG﹕ *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
11-17 17:00:18.693    3829-3829/? A/DEBUG﹕ Build fingerprint: 'google/angler/angler:6.0/MDB08L/2343525:user/release-keys'
11-17 17:00:18.694    3829-3829/? A/DEBUG﹕ Revision: '0'
11-17 17:00:18.694    3829-3829/? A/DEBUG﹕ ABI: 'arm64'
11-17 17:00:18.694    3829-3829/? A/DEBUG﹕ pid: 25264, tid: 25321, name: GpuThread  >>> io.wizkers.safecast.drive <<<
11-17 17:00:18.694    3829-3829/? A/DEBUG﹕ signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x440
11-17 17:00:18.723    3829-3829/? A/DEBUG﹕ x0   00000000000006d8  x1   0000000000000000  x2   0000000000000000  x3   0000000000000001
11-17 17:00:18.724    3829-3829/? A/DEBUG﹕ x4   0000000000000010  x5   0000000000000000  x6   0000000000000010  x7   0000000000000007
11-17 17:00:18.724    3829-3829/? A/DEBUG﹕ x8   00000000ffffffff  x9   0000000000000010  x10  00000000ffffffff  x11  0000000000000000
11-17 17:00:18.724    3829-3829/? A/DEBUG﹕ x12  0000000000000000  x13  0000000000000028  x14  0000000000000078  x15  0000007f71520780
11-17 17:00:18.724    3829-3829/? A/DEBUG﹕ x16  0000007f71520030  x17  0000000000000003  x18  0000000000000001  x19  0000000000000780
11-17 17:00:18.724    3829-3829/? A/DEBUG﹕ x20  0000007f716ba000  x21  0000000000000003  x22  0000007f71564000  x23  0000007f71567000
11-17 17:00:18.724    3829-3829/? A/DEBUG﹕ x24  0000007f71567cb8  x25  0000000000000001  x26  0000000000000000  x27  0000007f9a7bc140
11-17 17:00:18.724    3829-3829/? A/DEBUG﹕ x28  0000007f7167edfc  x29  0000007f7167eb50  x30  0000007fa0666024
11-17 17:00:18.724    3829-3829/? A/DEBUG﹕ sp   0000007f7167eb50  pc   0000007fa07195c4  pstate 0000000060000000
11-17 17:00:18.738    3829-3829/? A/DEBUG﹕ backtrace:
11-17 17:00:18.739    3829-3829/? A/DEBUG﹕ #00 pc 000000000024d5c4  /vendor/lib64/egl/libGLESv2_adreno.so (EsxGfxMem::UpdateTimestampBucketId(EsxContext const*, unsigned int, EsxAccessType)+64)
11-17 17:00:18.739    3829-3829/? A/DEBUG﹕ #01 pc 000000000019a020  /vendor/lib64/egl/libGLESv2_adreno.so (EsxContext::BucketRenderingCmds()+596)
11-1
```

randdusing, 2015-11-18:
Interesting, and it happens only when you've previously connected to the Bluetooth device within the app?

elafargue, 2015-11-18:
Unfortunately (for me) it looks like this is a much deeper Phonegap issue
on the Nexus 6P - the crash happens on the Chromium webview and is related
to the OpenGL libraries, so nothing to do with your plugin. It just often
happens right after I connect, because there's a lot of canvas action going
on. Impossible to reproduce the crash reliably, and since this is a bus
error Segfault - signal 11 - happening at very low level, not much i can do
to debug further, I'm afraid.

   I'm currently looking at using Crosswalk as the webview, but this makes
the APK super heavy...

closed 2015-12-23 by randdusing

johnnagro, 2016-02-22:
@elafargue i'm experiencing the same exact issue with a Chrome Webview (unrelated to PhoneGap) on a Nexus 6P. 

Did you ever resolve this or find more information?

elafargue, 2016-02-22:
@johnnagro : was tagged as "will not fix" by the Phonegap team - the issue is with the GPU driver on the Nexus 6P somewhere in OpenGL, so good luck getting someone to move and take ownership on that one..

The only workaround I have found so far: uninstall any Webview update you have on your Nexus 6P, the stock Android webview does not have the problem, it was introduced by a "bug fix" release later on.

Second possible workaround that you could consider: use Crosswalk as your webview rather than the stock Android webview, it works fine but adds significant bulk to your APK. Might not be suitable for your particular use since you are not using Phonegap.