randdusing/cordova-plugin-bluetoothle #209: When device disconnects or gets out of range, the wrong callback is called

elafargue, 2015-11-17:
When a device disconnects, rather that getting a "disconnected" status in the connectSuccess callback, I get a connection error (error: "connect", message: "connection failed") in the connectError callback: this is not what the documentation indicates, and makes device disconnects difficult to handle...

elafargue, 2015-11-17:
This is on all devices & android versions I tested (5.X and 6.0), Nexus 5,7,10,6P

randdusing, 2015-11-17:
Are you sure it's ever actually connected? The connectError callback is only used if the connection fails for some reason. Do you experience the same behavior on the sample app?

elafargue, 2015-11-17:
Yes, it is definitely connected. I will try on the sample app and report.

elafargue, 2015-11-17:
Confirmed on sample app as well: here is the log. I connected, waited a couple of seconds then switched off the bluetooth device. After about 15 seconds, the "error" callback was called as you can see:

```
Connect : {"address":"00:07:80:39:36:E4"}
index.js:1125 Connect Success : {"status":"connecting","address":"00:07:80:39:36:E4","name":"BLEBee 803936E4"}
index.js:1125 Connecting
index.js:1125 Connect Success : {"address":"00:07:80:39:36:E4","name":"BLEBee 803936E4","status":"connected"}
index.js:1125 Connected
index.js:1125 Connect Error : {"address":"00:07:80:39:36:E4","name":"BLEBee 803936E4","error":"connect","message":"Connection failed"}
```

Note that as long as I manage my disconnects/ reconnects through this callback, the app works absolutely great (thanks for the plugin, by the way!). Check out Safecast:Drive on the play store if you want to give it a spin...

randdusing, 2015-11-17:
Ok, let me take a look at this. I could have sworn it will throw a disconnected status using my heart rate monitor, but I may be wrong.

randdusing, 2016-01-25:
I was able to recreate this. This will be added to the version coming out within a day or two.

randdusing, 2016-01-25:
I believe this should be fixed in v3 with some changes to the scan locking mechanism. Let me know if it still exists.

closed 2016-01-25 by randdusing