randdusing/cordova-plugin-bluetoothle #211: Problem with the example

Juanca3893, 2015-11-17:
Hello randdusing, I think that i am having problems with the example.
I copy the www folder in my new project and I add the plugin "https://github.com/randdusing/BluetoothLE" and the "android" platform.
Then I build the project and I run that in my Android Device and the APP appears.

So when i click on Initialize my Bluetooth start to run. Then I click on "Start Scan" and I found a device whose name is "null" and i have others 2 Android devices more with Bluetooth activated, but they are not detected.

What am i doing wrong? What is the objective of the buttoms: "Is initialized" "Is enabled" and "Is Scanning"?

PD: Sorry if my english is bad, I am spanish.

Great work and thanks!

randdusing, 2015-11-17:
Sounds like you have the example working. Sometimes my address picks up "random" devices with null names. I'm not sure if it's an issue with the plugin or something lower level on Android. I never see those device when I'm scanning with my iPhone at the same time.

On the other devices you aren't see. Are you sure they are advertising in Bluetooth LE? If it's just a normal Bluetooth device, it won't show up.

Juanca3893, 2015-11-20:
OK, how can i know if my devices are Bluetooth LE? I am using Xperia Z1 and LG G2, but it doesn't work correctly.
Other question, how can i watch the logs? I write in cmd: adb logcat, but there are a lot of messages.

Thanks.

randdusing, 2015-11-20:
Those devices most likely support Bluetooth LE, but they won't be actively advertising as Bluetooth LE devices unless there's an app on the device doing so.

I use Android Device Monitor to monitor the logs and filter out only things pertaining to my app / plugin.

Juanca3893, 2015-11-20:
Thanks, now i can see the logs. The only strange log is:

Juanca3893, 2015-11-20:
Thanks for that quick answer, now i can see the logs and i will try to connect two devices and to find problems.
Can i use this example in two different devices (with BLE) and use it to send data(like text) from device1 and from device2 between them?

I don't know what the example does exactly, so i am a little lost.

Sorry for this silly questions and Thanks!

randdusing, 2015-11-20:
This plugin only implements the central (server) role. So it could read data from another device, but that other device wouldn't be able to use this plugin to send data. I'm planning to add client (peripheral) role support to the plugin, but not sure when. Additionally, if both your devices support normal Bluetooth, I don't think there's a benefit to using Bluetooth LE. I believe traditional Bluetooth can send data much quicker.

closed 2015-12-23 by randdusing