randdusing/cordova-plugin-bluetoothle #213: Write problem in characteristics

jandureu, 2015-11-23:
Hello,

I have a problem writing some data into a service characteristic.
I'm using LightBlue on an iPad to simulate my bluetooth device.

Everything is working except the "write" function.
My characteristic has "read" and "write" properties enabled. (I also tested with "write Without Response")

Here is the error :
```
Write Error : {"serviceUuid":"1816","name":"iPad","error":"write","message":"Error Domain=CBATTErrorDomain Code=3 \"Writing is not permitted.\" UserInfo={NSLocalizedDescription=Writing is not permitted.}","address":"23AA66A8-17A5-9C91-2B4F-7A2591E4BA84","characteristicUuid":"a1cd3767-6f55-4f62-ba53-2263a4eedd1b"}
index.js (776,20)
```
Do you have any idea ?

Thank you.

randdusing, 2015-12-23:
It looks like a permissions issue. I want to say I read something in the iOS documentation about needing to set a write permission property even if you set the characteristic to writeable. When I revisit the "server role" code, I can take another look.

RickyRakhmatP, 2016-01-22:
Hello,

I have same a problem writing some data into a service characteristic.

Here is code testing : 
 var string="Hello World Test Ricky Test ricky \n Test Tess Tes test \n Test \n test7 test 8 t \n";
 string += "Print From iOS \n ";

var bytes=bluetoothle.stringToBytes(string);
var value= bluetoothle.bytesToEncodedString(bytes);
write(address, serviceUuid, characteristicUuid, value);

Here is the error : 

Write Error : {"serviceUuid":"e7810a71-73ae-xxx-xxxx-xxxxxxxxxxx","name":"MTP_3-4","error":"write","message":"Error Domain=CBATTErrorDomain Code=13 \"The Value's length is invalid.\" UserInfo=0x1765d720{NSLocalizedDescription=The Value's length is invalid.}","address":"EC89F7BC-XXXX-XXXX-XXXX-XXXXXXXXXXXXXX","characteristicUuid":"bef8d6c9-xxxx-xxxx-xxxx-xxxxxxxxxxx"}

But , if string line 2 hide..print success..

iam testing to print bluetooth thermal printer : Model : MTP-3-BT-A (MP8020 I)

Please for help 

Thank you

randdusing, 2016-01-22:
@RickyRakhmatP Try splitting it up into operations of 20 bytes.

RickyRakhmatP, 2016-01-25:
you have example code to splitting operations of 20 bytes

Thank you

randdusing, 2016-01-25:
I don't have example code, but it should be pretty straightforward. Take the array of bytes from bluetoothle.stringToBytes and split into chunks of up to 20 byte. Then use bytesToEncoded string on each chunk and make a separate read request for each one

RickyRakhmatP, 2016-02-16:
after update plugin we have problem to write 

Write Error : {"name":"MTP_3-4","error":"characteristic","address":"EC89F7BC-1730-CE62-144B-0E92BEBB8853","message":"Characteristic not found"}

please help 

thank you

randdusing, 2016-02-16:
What's your code look like?

randdusing, 2016-03-31:
Anyone still having a problem?

zhuqingsong59, 2016-09-20:
when i use write() to send some message to bluetoothle device ,it response that 'Unable to write',did i local something?

randdusing, 2016-09-20:
@zhuqingsong59 Can you print out the discovery object for that service?

zhuqingsong59, 2016-09-20:
(1)function discoverSuccess(obj){
    // obj content   {"status":"discovered","address":"88:4A:EA:8F:68:FA","name":"TEST_BLUETOOTH",
    //     "services":[  {"uuid":"1801","characteristics": a object list}, {"uuid":"1800","characteristics": a object list},  {"uuid":"fff0","characteristics": a object list}, {"uuid":"180a","characteristics": a object list} ]}
    app.serviceUUID=obj.services[0].uuid;
    app.characteristicsUUID=obj.services[0].characteristics[0].uuid;
}
(2)sendElement.onclick=function(){
            var string = "Hello World";
            var bytes = bluetoothle.stringToBytes(string);
            var encodedString = bluetoothle.bytesToEncodedString(bytes);
            bluetoothle.write(writeSuccess, writeError,{ value:encodedString, service:app.serviceUUID,  characteristic:app.characteristicsUUID,  type:"noResponse", address:app.address }); }
(3)function writeError(obj){
    //obj content
    // { "address":"88:4A:EA:8F:68:FA", "name":"TEST_BLUETOOTH", "service":"1801", "characteristic":"2a05",  "error":"write", "message":"Unable to write" }
}
this is main process that i use ,did i make some mistake?

randdusing, 2016-09-21:
Your discover object should have a properties object. Check to make sure you're able to write without response. You may need to just remove the type parameter from your write call.

closed 2016-09-21 by randdusing

qq455158249, 2017-03-13:
if you send order use CBCharacteristicWriteWithResponse  , change the CBCharacteristicWriteWithoutResponse try again