randdusing/cordova-plugin-bluetoothle #214: Automatic reconnect

maximilianloy, 2015-11-24:
Hello,

I need to have the phone automatically reconnect to the device after a disconnect and when it is within reach again. At least the "connectGatt" method on Android comes with a autoConnect parameter, but this is hardcoded to "false" in the plugin.

I am wondering if there is a reason for not supporting this parameter in Javascript? To me it looks like exactly the behaviour I need. Does iOS support that behaviour too?

randdusing, 2015-11-24:
You could call the reconnect function when a disconnected event is receive.

maximilianloy, 2015-11-24:
If it disconnected because the phone and device got too far away, will the reconnect function make sure to connect as soon as they are within reach again?

randdusing, 2015-11-24:
I haven't verified the specific behavior, but it should. Calling the plugins reconnect method calls this internally on Android: http://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#connect() while iOS just uses the standard connect method. Neither of them timeout and should automatically connect when the device is within range again. Based on what I've seen with Android so far, I prefer to completely disconnect and close the connection, wait a small timeout and then call connect again rather than reconnect.

maximilianloy, 2015-11-24:
Sounds great, thanks!
If you completely disconnect and close the connection, does it require to re-discover the services and characteristics? What is the "cost" of closing the connection instead of just re-connecting?

randdusing, 2015-11-24:
iOS must be discovered on reconnect according to what I added to the README a while back, but I haven't reverified this lately. I can't say exactly what the cost of closing the connection is. The Android Bluetooth stack can be pretty buggy, so I prefer to completely close out the connection. I would test it out and see what works best for you. Just keep in mind that Bluetooth "reliability" seems to vary between Android devices, which is partially why you'll see so many issues. :)

closed 2015-12-23 by randdusing