randdusing/cordova-plugin-bluetoothle #216: NullPointerException crashes app on Android

epoplavskis, 2015-11-25:
Hi, 

I'm experiencing abrupt app crash during continues device scanning. Here is log from the console: 

```
11-25 17:38:15.904 25752-29964/com.richmond.chip D/BluetoothAdapter: startLeScan(): null
11-25 17:38:15.924 25752-25765/com.richmond.chip D/BluetoothAdapter: onClientRegistered() - status=0 clientIf=5
11-25 17:38:15.944 25752-28272/com.richmond.chip W/dalvikvm: threadid=11: thread exiting with uncaught exception (group=0x41ec4ce0)
11-25 17:38:16.014 25752-28272/com.richmond.chip E/AndroidRuntime: FATAL EXCEPTION: pool-1-thread-4
                                                                   Process: com.richmond.chip, PID: 25752
                                                                   java.lang.NullPointerException
                                                                       at com.randdusing.bluetoothle.BluetoothLePlugin.discoverAction(BluetoothLePlugin.java:1078)
                                                                       at com.randdusing.bluetoothle.BluetoothLePlugin.access$1000(BluetoothLePlugin.java:40)
                                                                       at com.randdusing.bluetoothle.BluetoothLePlugin$11.run(BluetoothLePlugin.java:383)
                                                                       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
                                                                       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
                                                                       at java.lang.Thread.run(Thread.java:841)
11-25 17:38:16.454 25752-25765/com.richmond.chip D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=8C:8B:83:53:D6:05
11-25 17:38:16.735 25752-25752/com.richmond.chip I/chromium: [INFO:CONSOLE(1485)] "AntiLostBLE - ", source: file:///android_asset/www/js/application.js (1485)
11-25 17:38:16.945 25752-25752/com.richmond.chip I/chromium: [INFO:CONSOLE(1485)] "AntiLostBLE - ", source: file:///android_asset/www/js/application.js (1485)
11-25 17:38:17.536 25752-28377/com.richmond.chip D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=8C:8B:83:53:D6:05
11-25 17:38:17.716 25752-25752/com.richmond.chip I/chromium: [INFO:CONSOLE(1485)] "AntiLostBLE - ", source: file:///android_asset/www/js/application.js (1485)
11-25 17:38:17.806 25752-25752/com.richmond.chip W/ScreenOrientationListener: Removing an inexistent observer!
11-25 17:38:17.806 25752-25752/com.richmond.chip E/chromium: [ERROR:runtime_javascript_dialog_manager.cc(69)] Not implemented reached in virtual void xwalk::RuntimeJavaScriptDialogManager::ResetDialogState(content::WebContents*)
11-25 17:38:17.926 25752-25752/com.richmond.chip W/ManagedEGLContext: doTerminate failed: EGL count is 2 but managed count is 1
11-25 17:38:28.838 25752-25867/com.richmond.chip D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=8C:8B:83:53:D6:05
11-25 17:38:30.069 25752-25766/com.richmond.chip D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=8C:8B:83:53:D6:05
11-25 17:38:30.069 25752-25752/com.richmond.chip W/ManagedEGLContext: doTerminate failed: EGL count is 2 but managed count is 1
11-25 17:38:39.859 25752-28272/com.richmond.chip I/Process: Sending signal. PID: 25752 SIG: 9
```

This occurred on HTC min one running Android 4.4.4. Would really appreciate your help debugging/fixing this.

randdusing, 2015-11-25:
Is this issue occurring on any other devices? And does it happen consistently?

It looks like the connection object is null, or the discoveredState property of the connection object is null. Most likely the former. Did you happen to call discover() about the same time as the device happen to drop its connection? It may be a race condition of some sorts on the connection object. If you can recreate it, I'll add a mutex around the connection as soon as possible and get you to retest.

randdusing, 2015-11-25:
Oh yah, I just noticed you talked about scanning, but the code in the log involves discovery(). Can you share your code, or at least the order of Bluetooth related calls.

epoplavskis, 2015-11-25:
Thanks for response.  I did some more testing on a nexus 5 running android 6.0 and was able to narrow down the issue. it happens when already connected ble device goes out of range and then comes back in. My app when device goes out of range starts a scan process looking for the lost device, once the scan returns the correct device, it calls reconnect function. Now this is where the issue happens, when reconnect error handler returns with ‘isNotDisconnected’, my app assumes that device is connected and proceeds to discover the device services and fatally crashes. It doesn’t happen like such every single time, which made it difficult to catch the issue, but i guess there is a way how ‘isNotDisconnected’ state should be handled in the app.

randdusing, 2015-11-25:
How do you when the device is lost? Are you receiving a "disconnected" status from the original connect callback? I'm trying to figure out how the isNotDisconnected error is possible in this situation since the device should have been disconnected before the scan and reconnect. Either way you may try adding a small timeout before the disconnect to fix any issues.

I looked at some other crash reports from my app and it looks like there have been some similar issues over the past few months. I'll try to add some mutex locks in a future release and see if that resolves the issue.

epoplavskis, 2015-11-26:
My app periodically reads some values of ble device, at the point when ble device is out of range, the read function returns an error '{error: "disconnected"}', as well as reconnect function triggers an error handler with message ' {error: "connect", message: "Connection failed"}'. both of handlers trigger app global event, which then starts the scan. 

I'll try adding some timeouts, though is there i way how should i handle the 'isNotDisconnected' error? Do you think it would help if i trigger disconnect function at the point when app thinks ble tag is lost?

randdusing, 2015-11-26:
Best bet is probably closing on the disconnect even and then calling connect again.

epoplavskis, 2015-11-27:
ok, i was able to resolve this by running disconnect function, when the app thinks device was lost, and just before connecting again to device.

epoplavskis, 2015-11-27:
Thanks for the help.

randdusing, 2015-11-28:
Glad, you figured it out. I'll keep this open to remind me about implementing a lock to ideally reduce these occurrences.

randdusing, 2016-03-31:
Current master runs everything on the WebCore thread, so there shouldn't be any race conditions causing these errors anymore.

closed 2016-03-31 by randdusing