randdusing/cordova-plugin-bluetoothle #218: Cannot subscribe to Characteristic on Android

joshdavidson613, 2015-12-04:
Hi,

I am having a major problem as I cannot subscribe to a characteristic and constantly get an error.

I use the following steps:

Scan
Connect
Discover
Subscribe

Here is the debug output and code I execute

startScan(); -> Start Scan Success : {"address":"BB:A0:50:10:11:12","name":"ArkrayBGM","rssi":-97,"advertisement":"AgEGCglBcmtyYXlCR00F/zEBOwQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=","status":"scanResult"}

connect("BB:A0:50:10:11:12"); -> Connect Success : {"address":"BB:A0:50:10:11:12","name":"ArkrayBGM","status":"connected"}

connect("BB:A0:50:10:11:12"); -> Connect Success : {"address":"BB:A0:50:10:11:12","name":"ArkrayBGM","status":"connected"}
---------------------------------
discover("BB:A0:50:10:11:12") ->
Discover Success : {  
   "status":"discovered",
   "address":"BB:A0:50:10:11:12",
   "name":"ArkrayBGM",
   "services":[  
      {  
         "serviceUuid":"1800",
         "characteristics":[  
            {  
               "characteristicUuid":"2a00",
               "properties":{  
                  "read":true
               },
               "descriptors":[  

               ]
            },
            {  
               "characteristicUuid":"2a01",
               "properties":{  
                  "read":true
               },
               "descriptors":[  

               ]
            },
            {  
               "characteristicUuid":"2a04",
               "properties":{  
                  "read":true
               },
               "descriptors":[  

               ]
            }
         ]
      },
      {  
         "serviceUuid":"1801",
         "characteristics":[  
            {  
               "characteristicUuid":"2a05",
               "properties":{  
                  "read":true,
                  "indicate":true
               },
               "descriptors":[  
                  {  
                     "descriptorUuid":"2902"
                  }
               ]
            }
         ]
      },
      {  
         "serviceUuid":"0003cdd0-0000-1000-8000-00805f9b0131",
         "characteristics":[  
            {  
               "characteristicUuid":"0003cdd1-0000-1000-8000-00805f9b0131",
               "properties":{  
                  "read":true,
                  "notify":true
               },
               "descriptors":[  
                  {  
                     "descriptorUuid":"20003d9c-0000-0000-0000-000020002903"
                  },
                  {  
                     "descriptorUuid":"20003d9c-0000-0000-0000-000020002904"
                  }
               ]
            },
            {  
               "characteristicUuid":"0003cdd2-0000-1000-8000-00805f9b0131",
               "properties":{  
                  "writeWithoutResponse":true
               },
               "descriptors":[  
                  {  
                     "descriptorUuid":"20003d9c-0000-0000-0000-000020002905"
                  }
               ]
            }
         ]
      }
   ]
}
---------------------------------
 var msg = new Uint8Array(1);
    msg[0] = 0x80;
    write("BB:A0:50:10:11:12", "0003cdd0-0000-1000-8000-00805f9b0131", "0003cdd2-0000-1000-8000-00805f9b0131", bluetoothle.bytesToEncodedString(msg), "noResponse"); ->     Write Success : {"address":"BB:A0:50:10:11:12","name":"ArkrayBGM","serviceUuid":"0003cdd0-0000-1000-8000-00805f9b0131","characteristicUuid":"0003cdd2-0000-1000-8000-00805f9b0131","status":"written","value":"gA=="}
---------------------------------
  subscribe("BB:A0:50:10:11:12", "0003cdd0-0000-1000-8000-00805f9b0131", "0003cdd1-0000-1000-8000-00805f9b0131"); ->
Subscribe : {"address":"BB:A0:50:10:11:12","serviceUuid":"0003cdd0-0000-1000-8000-00805f9b0131","characteristicUuid":"0003cdd1-0000-1000-8000-00805f9b0131"}

Subscribe Error : {"error":"descriptor","message":"Descriptor not found","address":"BB:A0:50:10:11:12","name":"ArkrayBGM"}
--------------------------------------

So I can scan, find, connect, discover, write, and read, but I cannot subscribe. The error is related to a "Descriptor not found", but the subscribe call doesn't ask for descriptor uuid only characteristic information.  What is going on?

randdusing, 2015-12-23:
The Client Configuration (00002902-0000-1000-8000-00805f9b34fb) descriptor has to be written to enable notification. Is this a customer characteristic?

randdusing, 2016-01-16:
Hi Josh, anymore information on this issue?

closed 2016-01-24 by randdusing